projecthorus/radiosonde_auto_rx #289: SoapySDR Integration

alexFab, 2020-07-09:
As far as i see, radiosonde_auto_rx force me to use RTL devices. 
It would be great if i could also use other devices via soapySDR.

darksidelemm, 2020-07-09:
Sure would be nice, but we need drop-in replacements for rtl_fm and rtl_power for this to happen.

rx_tools was one possible option, but it's extremely buggy and is unmaintained. If you can find something that is a feature-parity drop-in solution for both rtl_fm and rtl_power I'm all ears, otherwise there's not much that can be done without a significant re-write, and I do not have time for that.

darksidelemm, 2020-07-09:
A similar issue opened previously is here: https://github.com/projecthorus/radiosonde_auto_rx/issues/38

YO3IBZ, 2020-12-29:
It will be nice to have option to use different sdr's, not just rtl :D

darksidelemm, 2020-12-29:
As discussed in other issues, this is a huge amount of work that I don't really have the time to work on... If someone else would step up to develop something that can drop-in to place of rtl_fm and rtl_power that would be great...

sv1, 2022-06-18:
Check this out:

https://github.com/rxseger/rx_tools

darksidelemm, 2022-06-18:
This utility has already been looked into. It's abandoned, and is extremely buggy.

closed 2024-11-16 by darksidelemm