projecthorus/radiosonde_auto_rx #294: Detect WSL from Python in Windows

happysat, 2020-09-06:
Is it possible to add this to Auto-RX:

Function detects if Python is running in WSL.

import uname

def in_wsl() -> bool:
    """
    WSL is thought to be the only common Linux kernel with Microsoft in the name.
    """

    return 'Microsoft' in uname().release

Example overhere https://www.scivision.dev/python-detect-wsl/

To skip the lsusb checks?

darksidelemm, 2020-09-06:
Yep, we already skip those checks for OSX, so it shouldnt be difficult to add in a check for WSL too.

darksidelemm, 2020-09-19:
The testing branch now has a check for non-native Linux systems. Could you please give it a go and see if this works within WSL?

darksidelemm, 2020-09-19:
So detecting WSL isn't enough here. Since USB access is not possible from within WSL, the windows rtl-sdr binaries have been used via a symlink, and their behaviour don't appear to pass through return codes correctly. 
I'm loathe to make big changes to the code to try and work around this, so I'm closing this for now. If WSL ever supports direct USB access it might be possible to make auto_rx work reliably, without bypassing a heap of checks.

closed 2020-09-19 by darksidelemm