projecthorus/radiosonde_auto_rx #303: Provide dynamic KML

drid, 2020-10-03:
Provide kml file with autorefresh that can be use by GoogleEarth or other KML compatible apps

KML file could be something like 
```
           "<?xml version=\"1.0\" encoding=\"UTF-8\"?> \
            <kml xmlns=\"http://earth.google.com/kml/2.1\"> \
                <NetworkLink> \
                    <Link> \
                        <href>" + flask.request.host_url + "rs_feed.kml</href> \
                            <refreshMode>onInterval</refreshMode> \
                            <refreshInterval>10</refreshInterval> \
                    </Link> \
                </NetworkLink> \
            </kml>"
```

and **rs_feed.kml** can have the actual coordinates

darksidelemm, 2020-10-03:
I actually wrote something like this a looooong time ago, as a bit of a precursor to chasemapper ( https://github.com/projecthorus/chasemapper ) 
The code is here: https://github.com/projecthorus/horus_utils/tree/master/apps/kml_chase

Something like this (though without the predictions... that's getting a bit much to add into this repo) could probably be integrated into auto_rx if desired, and could hang off the backend of the existing web interface easily enough. 

Not sure I have the time to work on this in the short term though, it's a fairly big addition.

drid, 2020-10-03:
Should i give it a try?

darksidelemm, 2020-10-03:
Yep, that would be great! Make sure you target the testing branch.

drid, 2020-10-06:
Still a WIP currently under testing https://github.com/projecthorus/radiosonde_auto_rx/pull/304

darksidelemm, 2020-12-28:
Closed in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.4.0

closed 2020-12-28 by darksidelemm