projecthorus/radiosonde_auto_rx #307: Timezone Not The Same in Log?

dh47, 2020-10-19:
Last night, NZD time witch is GMT +13 right now at around 10:34pm the balloon launched.

The pi's timezone is correct with the NZ time set, but in the log, I see this.

`2020-10-19T09:34:31.000Z,S0461107,1963,-36.79947,174.63147,683.2,4.8,14.2,103.6,-273.0,-1.0,RS41,403.001,17.2,0,10,3.0,-1,-1`

As you can see its alittle out,  Is there a way to found out why?  I have got a setting incorrect?

darksidelemm, 2020-10-19:
The timestamps in the log files are not local time. Instead, they are the time and date as reported by the radiosonde itself. This is always UTC time, as indicated by the 'Z' at the end of the timestamp.

Does this answer your question?

Cheers,
Mark

dh47, 2020-10-19:
Oh I see ok thanks, yes that makes sense.  I'll just have to somehow take
that into account.

Not sure how to close via email.

closed 2020-10-19 by darksidelemm