projecthorus/radiosonde_auto_rx #309: UDP/JSON input format needs to be documented

darksidelemm, 2020-10-26:
Currently we use a JSON input format to pass data from the various RS decoders into auto_rx. There are lots of fields, and some are handled differently depending on the sonde type. 
The fields, and how they are used, needs to be documented better. 

Write a wiki page!

rs1729, 2020-11-05:
Here are the JSON types and subtypes of the current decoders:

```
Vaisala RS92-SGP, RS92-NGP
RS92 = { "type": "RS92", "id": "..." }
 (for L-band RS92-NGP option '--ngp' gives the right internal frequency)

Vaisala RS41
RS41 = { "type": "RS41", "id": "...", subtype: "..." }
subtypes: "RS41-SG", "RS41-SGP", "RS41-SGM", "RS41-NG"

Meteomodem M10, M10+
M10  = { "type": "M10", "id": "M10-...", "subtype": "0x9F" }
M10+ = { "type": "M10", "id": "M10-...", "subtype": "0xAF" }

Meteomodem M20
M20  = { "type": "M20", "id": "M20-...", "subtype": "0x20" }

Graw DFM-06, DFM-09, DFM-17, PS-15
DFM-06 = { "type": "DFM", "id": "DFM-...", "subtype": "0x6" }
DFM-09 = { "type": "DFM", "id": "DFM-...", "subtype": "0xA" }
other subtypes:
"0xB": DFM-17 ?
"0xC": DFM-09P or DFM-17 ?
"0xD": DFM17 ?
"0x7", "0x8": PS-15 ?

Lockheed Martin Sippican LMS6-403, LMSX
LMS6-403 = { "type": "LMS", "id": "LMS6-8..", "subtype": "6" }
LMSX     = { "type": "LMS", "id": "LMSX-...", "subtype": "X" }

Lockheed Martin Sippican LMS6-1680/MK2A
LMS6-1680/MK2A = { "type": "LMS", "id": "LMS6-7..", "subtype": "MK2A" }

Meisei iMS-100
iMS-100 = { "type": "IMS100", "id": "IMS100-..." }

Intermet iMet-1-RS, iMet-4
iMet-1-RS, iMet-4 = { "type": "IMET", "id": "iMet" }

Meteolabor SRC-C50
C50 = { "type": "C50", "id": "C50-..." }
```
Aternatively, for Meisei iMS-100 one could think of { "type": "MEISEI", "id": "IMS100-...", "subtype": "IMS100" }, but only for iMS-100 there is JSON output.

darksidelemm, 2020-11-06:
At the moment the only subtype that I'm using are the ones from the RS41, and the DFM (which i decode to the model). The rest I ignore.

I'll look into adding additional subtype data in, though at the moment the beginning of the serial number is usually enough to discriminate by different types (except LMS6 400/1680 MHz, but the frequency gives those away).

rs1729, 2020-11-23:
I changed the iMS-100 json type to {"type": "MEISEI", "subtype": "IMS100"}.
Now it is the same as the scan result, and if json output for Meisei RS11-G is included, it would be {"type": "MEISEI", "subtype": "RS11G"}.
@darksidelemm 
If you include the new meisei100mod.c, the commit
https://github.com/projecthorus/radiosonde_auto_rx/commit/cc0c4f1fea29b0f76088771a7b3406170acdf66f
in the testing branch would have to be rolled back. (You can still update the IMS100-xxxxxx comment.)

darksidelemm, 2020-12-28:
Documentation here: https://github.com/projecthorus/radiosonde_auto_rx/wiki/RS----Auto_RX-JSON-Formats

closed 2020-12-28 by darksidelemm