projecthorus/radiosonde_auto_rx #311: Meisei telemetry type: 'IMS100'

rs1729, 2020-10-30:
In https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/auto_rx.py#L479,
`telemetry` is the decoder JSON string?
For Meisei IMS100,
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/meisei100mod.c#L769
returns "type": "IMS100". (Since IMS100 is the only supported Meisei radiosonde, there is no subtype and the type is "IMS100", not "MEISEI"; dft_detect does not differentiate, so it returns "MEISEI".)

darksidelemm, 2020-10-30:
Hmm, I wonder when that bug crept in. Im sure we were decoding iMS100 sondes at some point. 

Anyway, will change that line in the testing branch.

closed 2020-10-30 by darksidelemm

rs1729, 2020-10-30:
Yes, I have seen some IMS100 over Turkey.

rs1729, 2020-11-23:
https://github.com/projecthorus/radiosonde_auto_rx/issues/309#issuecomment-732407182
I changed the iMS-100 json type to {"type": "MEISEI", "subtype": "IMS100"}.
Now it is the same as the scan result, and if json output for Meisei RS11-G is included, it would be {"type": "MEISEI", "subtype": "RS11G"}.
@darksidelemm
If you include the new meisei100mod.c, the commit
cc0c4f1
in the testing branch would have to be rolled back. (You can still update the IMS100-xxxxxx comment.)