projecthorus/radiosonde_auto_rx #314: Decode LMS6 data

Reid-n0rc, 2020-11-16:
Add the ability to decode the LMS6 radiosonde data. Currently the software will decode the location information, but not the temperature and humidity.

I do not have additional information n the LMS6 data format right now. I am opening the issue to help with tracking and hopefully start some discussion on the LMS6 data format.

darksidelemm, 2020-11-16:
This requires detailed information on the LMS6 data format, and how the calculations to determine the temperature and humidity are performed. Then we need the same for both the 1680 MHz and 400 MHz variants.

I would be *very* surprised if LMS would give up this information.

darksidelemm, 2020-11-17:
I'm wondering if @rs1729 has any information on the above? Definitely a lot of LMS6 sondes in use in the US... what would be required to get temperature/humidity data out of it?

rs1729, 2020-11-17:
No, I didn't look into the data, the LMS6 is not flying here.
Temperature is usually easier than humidity. If it is a thermistor sensor, then you would probably see raw sensor data that changes exponentially (could be integer ADC values or float) below approx 12km altitude whereas the temperature changes roughly linearly (see the reported soundings). If you know the sensor, you could probably find the coefficients from the manufacturer. Also for thermistor calibration values are usually not so important.

bazjo, 2020-11-19:
https://www.graw.de/de/aktuelles/presse/graw-liefert-an-usa-noaa-national-weather-service/

With the transistion toward the 400 MHz Band in the US the time of LMS6 sondes will come to an end sooner than later. As far as I am aware most, if not all of the manned stations will be supplied by Graw, Vaisala probably will be continue to be present in the US as well. So probably the best thing to do with regard to US RX Stations is to improve handling of DFM-17s instead of adding an almost legacy sonde

darksidelemm, 2020-11-19:
Interesting information, thanks!

I know there are quite a few stations in the US that have switched from 1680 MHz LMS6 sondes to 400 MHz sondes, but it's interesting to know that they are transitioning to Graw sondes.

Reid-n0rc, 2020-12-02:
The only DFM sondes I have seen on the map recently have come out of Stillwater.  There were two yesterday 12/1. The other was out of Los Angeles 8 days ago.  If they're going to switch, they seem to only be in the testing stages.

darksidelemm, 2024-11-16:
Unsure if this really a priority any more. LMS6's are far less common nowdays. Still used at some rocket launch sites, but I suspect this will probably never happen.

closed 2024-11-16 by darksidelemm