projecthorus/radiosonde_auto_rx #315: dependencies - missing a package

agent-cooper, 2020-11-21:
add 'libtools' to the list of packages to be installed on raspberry pi

darksidelemm, 2020-11-21:
This isn't brought in as a dependency of any of the other packages? 
At what point does not having this package break the installation steps?

agent-cooper, 2020-11-21:
I was just installing on an RPi 3 with the latest Raspbian, and I got an error message saying that it was looking for libtoolize which is in the libtool package

darksidelemm, 2020-11-21:
Yes, but at which stage was this? During build of the rtl-sdr libraries, or somewhere else?

agent-cooper, 2020-11-21:
After entering: sudo apt-get install rtl-sdr

darksidelemm, 2020-11-21:
OK, that's extremely odd, since there should be no building going on at that point. No harm in adding libtool to the package list to be installed, but I haven't had any other reports of this being an issue.

closed 2020-11-21 by darksidelemm