projecthorus/radiosonde_auto_rx #318: Data not appearing on radiosondy.info when pushed

cjastone, 2020-11-28:
Hi team,

Apologies in advance if this isn't the correct way to raise this; I can't be certain it's not user error however I've tried to validate as much as I can, but there's not a lot of information out there on getting this set up to report to radiosondy.info.  Should I be successful in getting this up and running, I promise to document and post the solution on my blog for others to benefit from!

As best as I can gather, this seems to be something that should work out of the box however I've had no success.

I appear to be uploading packets as expected:

`Nov 28 11:20:00 pitwo auto_rx[11884]: 2020-11-28 11:20:00,464 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002015h3745.04S/14552.28EO028/001/A=066065 Clb=7.8m/s t=-56.2C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !wol!
Nov 28 11:20:30 pitwo auto_rx[11884]: 2020-11-28 11:20:30,411 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002045h3745.04S/14552.32EO047/005/A=066643 Clb=3.8m/s t=-57.1C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !wIh!
Nov 28 11:21:00 pitwo auto_rx[11884]: 2020-11-28 11:21:00,449 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002115h3744.98S/14552.37EO359/008/A=067208 Clb=7.3m/s t=-56.7C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !wTE!
Nov 28 11:21:30 pitwo auto_rx[11884]: 2020-11-28 11:21:30,507 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002140h3744.94S/14552.34EO311/010/A=067700 Clb=8.6m/s t=-55.4C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !w,q!
Nov 28 11:22:00 pitwo auto_rx[11884]: 2020-11-28 11:22:00,438 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002212h3744.93S/14552.22EO257/013/A=068310 Clb=3.1m/s t=-55.3C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !wqP!
Nov 28 11:22:30 pitwo auto_rx[11884]: 2020-11-28 11:22:30,387 INFO:APRS-IS - Uploaded to APRS-IS: CJAST2-14>APRARX,SONDEGATE,TCPIP,qAR,CJAST2-14:;R3770049 *002244h3745.00S/14552.13EO198/009/A=068925 Clb=8.9m/s t=-55.7C 401.501 MHz Type=RS41-SG Radiosonde auto_rx v1.3.2 !wm*!
`

Unfortunately these don't show on radiosondy.info.  I've downloaded the kxyTrack and tested with that, which does upload packets just fine, however in a slightly different format to the above:

`CJAST2-14>APLWS2:;R3770049 *233719h3747.11S/14504.11EO123/053/A=019347!wl:!Clb=6.2m/s t=-11.8C 401.50MHz Type=RS41-SG Details on http://radiosondy.info`

Any thoughts or suggestions?

Many thanks in advance and please keep up the amazing work!

darksidelemm, 2020-11-28:
Unsure... radiosondy.info's APRS packet parsing is a bit of a mystery to me, and I haven't been able to make contact with the developer to find out exactly what format I should be using. 
I do note that my station (and quite a few others) are uploading to radiosondy.info fine. It may be related to the way your callsign (which if going into the APRS-IS network should be a valid amateur radio callsign!) is used in the addressing fields. Apart from switching to a real callsign I'm unsure what else to suggest here.

cjastone, 2020-11-28:
Thanks Mark, appreciate the response!  Agree that others seem to not be having issues.

Regarding callsigns, I'm using the one generated by radiosondy.info's  kxyTrack image for now, just to keep it consistent and rule that out as a possible issue.  (Also the foundation license hasn't been sat as yet but that's definitely on the cards ASAP!)

If it's not a huge hassle, could I get you to review the below config against yours and see if anything obvious is different?  I note that I'm not sure exactly what should be in "aprs_pass" as the "aprs_user" field does not match my login for radiosondy.info - is "aprs_user" the correct field to be specifying my call sign?

[aprs]
aprs_enabled = True
aprs_user = CJAST2-14
aprs_pass = [account password for radiosondy.info?]
upload_rate = 30
aprs_server = radiosondy.info
aprs_object_id = <id>
aprs_position_report = False
aprs_custom_comment = Clb=<vel_v> t=<temp> <freq> Type=<type> Radiosonde
station_beacon_enabled = True
station_beacon_rate = 30
station_beacon_comment = radiosonde_auto_rx SondeGate v<version>
station_beacon_icon = /`

darksidelemm, 2020-11-28:
OK, in this case the issue is likely your aprs_pass field. The APRS-IS system requires a login using a callsign and a pass code, which is essentially a hash function for your callsign. It used to be that this hash function was a closely guarded secret, and you had to 'apply' for a passcode, but nowadays you can just use a bunch of online generators to calculate it. 

I suspect kxyTrack just generates a passcode itself, but I didn't really want to do that, to avoid bogus or mis-typed callsigns ending up in APRS-IS.

There's a suitable passcode generator here: https://apps.magicbug.co.uk/passcode/

Try using the passcode it generates there and see if it works.

cjastone, 2020-11-28:
Thanks Mark!  Appreciate the informative response.  I'll have to wait about 6 hours to know if it works, but will report back!

closed 2020-11-29 by cjastone