projecthorus/radiosonde_auto_rx #319: ImportError: cannot import name JSONDecodeError

ZL1LAC, 2020-11-29:
Never seen this error before!

```
pi@raspberrypi:~/radiosonde_auto_rx/auto_rx $ python auto_rx.py
Traceback (most recent call last):
  File "auto_rx.py", line 31, in <module>
    from autorx.web import start_flask, stop_flask, flask_emit_event, WebHandler, WebExporter
  File "/home/pi/radiosonde_auto_rx/auto_rx/autorx/web.py", line 23, in <module>
    from flask_socketio import SocketIO
  File "/usr/local/lib/python2.7/dist-packages/flask_socketio/__init__.py", line 21, in <module>
    import socketio
  File "/usr/local/lib/python2.7/dist-packages/socketio/__init__.py", line 3, in <module>
    from .client import Client
  File "/usr/local/lib/python2.7/dist-packages/socketio/client.py", line 7, in <module>
    import engineio
  File "/usr/local/lib/python2.7/dist-packages/engineio/__init__.py", line 3, in <module>
    from .client import Client
  File "/usr/local/lib/python2.7/dist-packages/engineio/client.py", line 2, in <module>
    from json import JSONDecodeError
ImportError: cannot import name JSONDecodeError
```

darksidelemm, 2020-11-29:
Not sure sorry! Is this from a fresh Rpi install?

ZL1LAC, 2020-11-29:
Yes, I'm trying this again and will report back.

ZL1LAC, 2020-11-29:
Reinstalled the pi os and followed guide, still getting the error.

darksidelemm, 2020-11-29:
I suspect it might be a Python 2 vs 3 issue. There may be a dependency that
now only supports Python 3.

Can you try using the relevant Python 3 packages? (E.g. python3-blah for
apt packages, and using pip3 for installing Python packages)

ZL1LAC, 2020-11-29:
Does this project support Python 3?

darksidelemm, 2020-11-29:
Sure does! I'm very close to dropping support for anything older than
Python 3.6

ZL1LAC, 2020-11-29:
Hmm ok I'll try that again using Python 3.  I assume the guide will be updated and some point with Python 3 changes?

darksidelemm, 2020-11-29:
Definitely on the To-Do list! I hadn't done it yet because Raspbian was still shipping Python2.7 by default...

Phils-git, 2020-12-01:
Hello ZL1LAC, did you have any joy with the errors..  I'm also trying to setup a new build and have the same problem. Have tried various Pi builds and been messing with it for several hours now with no luck, I did notice if you use...
python3 auto_rx.py  to do the initial testing you get a different result, but still no go.

Cheers.

ZL1LAC, 2020-12-01:
@Phils-git Yeah, so I did the following.

`sudo apt autoremove python-numpy python-setuptools python-crcmod python-requests python-dateutil python-pip sox git build-essential libtool cmake usbutils libusb-1.0-0-dev rng-tools libsamplerate-dev`

Then:

`sudo apt-get install python3-numpy python3-setuptools python3-crcmod python3-requests python3-dateutil python3-pip sox git build-essential libtool cmake usbutils libusb-1.0-0-dev rng-tools libsamplerate-dev`

And:

`sudo pip3 flask flask-socketio`

darksidelemm, 2020-12-01:
This looks about right. Unfortunately I'm not in a position to be able to run through this from scratch on a fresh RPi, so updating the installation guide will have to wait a few weeks.

Phils-git, 2020-12-01:
Arr Ha !
That's great work, I now have a system that that I can continue working on.

sudo pip3 install flask flask-socketio 
And when you start it you need to use python3 anywhere you see the word python, I now have systemd setup working and might have a look at some of the add on packages next.

You have made this little Vegemite very happy..  Thanks again from VK3 land..

:-)

ZL1LAC, 2020-12-01:
@Phils-git Are you also using systemd Service?   Don't forget to change the python location.

`ExecStart=/usr/bin/python3 /home/pi/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0`

pablosoria, 2020-12-04:
Same here, with a fresh Raspbian installation (v10 Buster).
With `python auto_rx.py` python 2.7.16  is executed and I obtain the error above.
After installing the packets for python 3 like @ZL1LAC said now it's working with: `python3 auto_rx.py
`
:)

ZL1LAC, 2020-12-06:
I would help with the wiki and edit it to the new python 3 commands, but no permission too.

darksidelemm, 2020-12-09:
I've now updated the instructions and run through them on a fresh Raspbian Lite installation. All seems to be working.

I've also updated the systemd and auto_rx.sh scripts to use python3.

closed 2020-12-09 by darksidelemm

Phils-git, 2020-12-09:
G'day,
no problems with the Python3 path and also anywhere I saw pip I used pip3.
One issue that did keep scratching for ages was the ppm cal setting in the
cfg file, as I found out...  it MUST be whole numbers, no decimal points,
when I entered a cal figure of 77.9 the gui failed to load however as soon
as I set it to 80, all was well again.

All seems to be working, actually recovered my first one today and even
managed to see it land which was a bit fancy.
I'm intrigued as to how the parachute works though, if in fact there is
one, all that I have left is a big mess of balloon rubber.

darksidelemm, 2020-12-09:
The PPM input is easy enough to fix :-) It'll be fixed in 1.3.3.

dterweij, 2021-01-27:
Hmmm... So with the 1.4.0  auto rx version, 2 systems are broken here (raspbian and ubuntu)

I have to decide:

- break 5 other tools that are running fine at python 2.7 and not at 3,x
- Revert the raspian version and keep running the old version/Don't do a git pull at the ubuntu system 
- stop with the balloon tracking hobby....

`pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo pip3 flask flask-socketio
sudo: pip3: command not found
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo update-alternatives --config python
There are 2 choices for the alternative python (providing /usr/bin/python).

  Selection    Path                Priority   Status
------------------------------------------------------------
  0            /usr/bin/python3.5   2         auto mode
* 1            /usr/bin/python2.7   1         manual mode
  2            /usr/bin/python3.5   2         manual mode

Press <enter> to keep the current choice[*], or type selection number: 0
update-alternatives: using /usr/bin/python3.5 to provide /usr/bin/python (python) in auto mode
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo pip3 flask flask-socketio
sudo: pip3: command not found
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo pip flask flask-socketio
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named 'pip._internal'
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sh startsonde.sh 
Traceback (most recent call last):
  File "./auto_rx.py", line 36, in <module>
    from autorx.web import (
  File "/home/pi/radiosonde_auto_rx/auto_rx/autorx/web.py", line 24, in <module>
    from flask_socketio import SocketIO
ImportError: No module named 'flask_socketio'
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo pip flask flask-socketio
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named 'pip._internal'
pi@repeater:~/radiosonde_auto_rx/auto_rx $ sudo pip3 flask flask-socketio
sudo: pip3: command not found
pi@repeater:~/radiosonde_auto_rx/auto_rx $ python --version
Python 3.5.3
`

@darksidelemm

darksidelemm, 2021-01-27:
To put it fairly bluntly - I can't continue to support older distributions. Python 2 has been out of support for a while now. If you have tools that don't work in Python 3, then you should be working with their developers to get them to support Python 3. Everyone has known Python 2 has been going out of support for a *long time* now.

That you are on Python 3.5 suggests you are on an older raspbian distribution. Current raspbian distributions provide Python 3.7.

darksidelemm, 2021-01-27:
Another question is is did you ever install python3-pip? If you can't run pip3 then it suggests you are missing that package, and by running pip you are only installing dependencies for Python 2.7 (though I don't know why pip is broken).

That you are running `sudo pip3 install flask flask-socketio` is also going to break other things. Take a look at the current instructions: https://github.com/projecthorus/radiosonde_auto_rx/wiki#31-software-dependencies which clearly specifies certain versions of flask-socketio, to avoid pip downloading flask-socketio v5, which breaks compatibility with the current web frontend.

darksidelemm, 2021-01-27:
Final note: You don't have to use the update-alternatives and switch everything to python 3 if you don't want. In all the instructions I give I explicitly specify the use of 'python3' and 'pip3' instead of 'python' and 'pip' which ensures python 3 is used, no matter what update-alternatives is set to.

snh, 2021-01-29:
Worth keeping in mind that deploying _radiosonde_auto_rx_ using Docker is an option if you have dependency challenges like this. Running _radiosonde_auto_rx_ inside Docker keeps it isolated from your existing Python 2 dependent applications, you don't even need Python installed on the host system in fact.

See https://github.com/projecthorus/radiosonde_auto_rx/wiki/Docker for instructions.

darksidelemm, 2021-01-29:
Using a venv would also solve this problem... though the systemd service fule would need to be updated to point to the venv's python binary instead of the system one.