projecthorus/radiosonde_auto_rx #320: Gain settings

Crille-ONCR16, 2020-12-04:
Hi,

What is the best Gian level when i use a lna 403mhz from Uptronics. Because on GQRX i see a high level on noise ...

Tx
Chris (ONCR16)

darksidelemm, 2020-12-04:
This is is a difficult question to answer as it depends on a lot of factors such as your antenna and your local noise environment.

I'd suggest just running with AGC for a while (-1) and seeing how the station performs.

closed 2020-12-04 by darksidelemm

rs1729, 2020-12-05:
It is often recommmended to start with 0 gain and then increase the gain until noise floor starts rising about 6-9 dB.
If you have a rs41 signal approx. 10-12 dB strong you can experiment how much gain is best for decoding, the error correction would give you the number of errors that have been corrected in each frame. Although it is difficult to find a static radiosonde signal that doesn't change with time.