projecthorus/radiosonde_auto_rx #325: Floating point exception

ZL1LAC, 2020-12-11:
Fresh install on a fresh pi.

```
pi@raspberrypi:~/radiosonde_auto_rx/auto_rx $ python3 auto_rx.py
2020-12-11 23:33:55,443 INFO:Reading configuration file...
2020-12-11 23:33:58,343 INFO:Config - Tested SDR #0 OK
 * Serving Flask app "autorx.web" (lazy loading)
2020-12-11 23:33:58,350 INFO:Started Flask server on http://0.0.0.0:5000
 * Environment: production
2020-12-11 23:33:58,353 INFO:Telemetry Logger - Started Telemetry Logger Thread.
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
2020-12-11 23:33:58,383 INFO:APRS-IS - APRS Uploader Started.
2020-12-11 23:33:58,385 INFO:OziMux - Started OziMux / Payload Summary Exporter
2020-12-11 23:33:58,646 INFO:Version - Local Version: 1.3.2  Current Master Version: 1.3.2
2020-12-11 23:33:58,646 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2020-12-11 23:34:01,137 INFO:Habitat - Station position uploaded.
2020-12-11 23:34:01,528 INFO:Scanner #0 - Starting Scanner Thread
2020-12-11 23:34:01,536 INFO:Scanner #0 - Running frequency scan.
2020-12-11 23:34:01,641 CRITICAL:Scanner #0 - rtl_power call failed with return code 136.
2020-12-11 23:34:01,642 CRITICAL:Scanner #0 - rtl_power reported error: Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 000002

Using device 0: Generic RTL2832U OEM
Floating point exception

2020-12-11 23:34:01,643 WARNING:Scanner #0 - RTLSDR produced no output... resetting and retrying.
2020-12-11 23:34:01,729 INFO:RTLSDR - Attempting to reset: Bus: 1  Device: 13
2020-12-11 23:34:11,970 INFO:Scanner #0 - Running frequency scan.
2020-12-11 23:34:12,080 CRITICAL:Scanner #0 - rtl_power call failed with return code 136.
2020-12-11 23:34:12,081 CRITICAL:Scanner #0 - rtl_power reported error: Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 000002

Using device 0: Generic RTL2832U OEM
Floating point exception

2020-12-11 23:34:12,082 WARNING:Scanner #0 - RTLSDR produced no output... resetting and retrying.
2020-12-11 23:34:12,167 INFO:RTLSDR - Attempting to reset: Bus: 1  Device: 13
2020-12-11 23:34:22,402 INFO:Scanner #0 - Running frequency scan.
2020-12-11 23:34:22,508 CRITICAL:Scanner #0 - rtl_power call failed with return code 136.
2020-12-11 23:34:22,509 CRITICAL:Scanner #0 - rtl_power reported error: Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 000002

Using device 0: Generic RTL2832U OEM
Floating point exception

2020-12-11 23:34:22,509 WARNING:Scanner #0 - RTLSDR produced no output... resetting and retrying.
2020-12-11 23:34:22,594 INFO:RTLSDR - Attempting to reset: Bus: 1  Device: 13
```

darksidelemm, 2020-12-11:
This error is coming from rtl_power itself, not from auto-rx. Turn on
debugging output (auto_rx.py -v), copy the rtl_power command that is run,
and try running it yourself and see if the error is still present.

ZL1LAC, 2020-12-11:
Hmmmmmmm.

```
pi@raspberrypi:~/radiosonde_auto_rx/auto_rx $ rtl_power -f 403000000:403000000:800 -i 20 -1 -c 20% -p 0 -d 0 log_power_0.csv
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 000002

Using device 0: Generic RTL2832U OEM
Floating point exception
```

darksidelemm, 2020-12-11:
Hold on, why are your start and stop frequencies the same? (both 403 MHz) That would likely result in bad things happening.

ZL1LAC, 2020-12-11:
Oh, I see it's likely to be a configuration error on my part, I did that because I wasn't getting some noise on either side which I thought might interfere with the process.

darksidelemm, 2020-12-11:
Give it a test, and if it works close the issue :-)

ZL1LAC, 2020-12-11:
Looks like it was a user error.  Closing this move on nothing to see here.  /me slaps himself with a smelly tuna fish.

varnav, 2021-08-29:
This cryptic error happens if `min_freq` and `max_freq` are the same. I think it worth adding correct error output.

darksidelemm, 2021-08-29:
... probably better to add some sanity checks when the configuration file is parsed, rather than trying to catch the error later.

shin19855, 2023-05-07:
I have the same issue when setting auto_rx in 'continuous' mode. I solved it by adding a **sudo** in the the **auto_rx.service** file: 

ExecStart=**sudo** /usr/bin/python3 /home/pi/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0

darksidelemm, 2023-05-07:
There should be no need to run auto_rx as sudo, and this is not recommended.

shin19855, 2023-05-11:
I was using the native installation methods, and followed exactly the steps on the Wiki page. The same error occurred twice, on two different sets of RPi + SDR. 

My knowledge on RPi and Linux is limited. So, I am all ears with a better solution.

darksidelemm, 2023-05-11:
If you are getting errors *without* running as sudo, then it's likely that you have run auto_rx as sudo at some point previously, and there are now files (log file outputs, other temporary tiles) that your user does not have permissions for.

closed 2024-11-16 by darksidelemm

DuoFold, 2025-04-01:
Ha! This one bit me too :) I set min and max freq the same, because I was targeting just one specific sonde.