projecthorus/radiosonde_auto_rx #326: M20 Error

oe3osb, 2020-12-13:
Hi, 
i faced out that the M20 are not walking to APRS. This is my Error: 

ERROR | No APRS ID conversion available for sonde type: M20

Any ideas how i can correct it ?

73 Erich

darksidelemm, 2020-12-13:
There is not yet an APRS ID conversion for M20 sondes, because there hasn't been any agreement on it between other radiosonde developers. Until this happens I won't be allowing M20 sondes to upload to APRS.
Uploading to tracker.sondehub.org will still work fine.

closed 2020-12-13 by darksidelemm

KosutM, 2021-02-09:
What about now ? Few of the receivers are currently uploading M20 data to APRS, so you can try to copy those packet format...
For example:
https://radiosondy.info/sonde_archive.php?sondenumber=ME8101554

darksidelemm, 2021-02-09:
I'm guessing this is from dxlAPRS? Is there any (readable) information on how this serial number is generated?

KosutM, 2021-02-10:
I don´t know, but auto_rx can decode M20, so I think there is no problem to add new elif for M20 here: radiosonde_auto_rx/auto_rx/autorx/aprs.py
Or am I missing something?

darksidelemm, 2021-02-10:
The serial number that is uploaded by dxlAPRS is generated from the binary format of the serial number, mashed up in some way to make it fit in an APRS object ID. 
What I get from the decoders is the serial number in the same format as is on the sticker on the box. Generating the 'aprs id' will need to be done at the binary level, so in the decoder itself. Doable, yes, but we need more information.

KosutM, 2021-02-10:
At least for uploading to radiosondy.info server, serial number as a sticker on the box should be enough, or not? Their parsers uses serial number format as is on the box.

darksidelemm, 2021-02-10:
No - information is uploaded to radiosondy.info via APRS-IS, which means it is limited to APRS packet formats. e.g. object names with only up to 9 characters.