projecthorus/radiosonde_auto_rx #328: Latest flask-socketio (v5) isn't compatible with socket.io-1.4.5.js

dbradby, 2020-12-15:
The current instructions install the latest version of flash-socketio which has now moved to version 5 and isn't compatible with the socket.io js being used (1.4.5) - https://flask-socketio.readthedocs.io/en/latest/

I would suggest changing the wiki from 

`sudo pip3 install flask-socketio`
to
`sudo pip3 install flask-socketio==4.3.2`

until the socket.io js is updated to 3.x

darksidelemm, 2020-12-15:
Good catch! Thanks for that. Will update the instructions.

Do you know if the new socket.io JS version will be backwards compatible?

dbradby, 2020-12-15:
It doesn't look backwards compatible according to https://socket.io/docs/v3/migrating-from-2-x-to-3-0/index.html

`TL;DR: due to several breaking changes, a v2 client will not be able to connect to a v3 server (and vice versa)`

and Flask-SocketIO has moved protocol versions. 

JavaScript Socket.IO version | Socket.IO protocol revision | Engine.IO protocol revision | Flask-SocketIO version | python-socketio version | python-engineio version
-- | -- | -- | -- | -- | --
0.9.x | 1, 2 | 1, 2 | Not supported | Not supported | Not supported
1.x and 2.x | 3, 4 | 3 | 4.x | 4.x | 3.x
3.x | 5 | 4 | 5.x | 5.x | 4.x

So you could move the JS to 3.x (I can give this a try this weekend) but existing installs would need to update their flask-socketio version as well. This is probably best done by moving that dependancy to a requirements.txt file or something and having it part of a usual update process. 

But for now `sudo pip3 install flask-socketio==4.3.2` will work

darksidelemm, 2020-12-15:
OK, I think we stick with pinning the flask-socketio version for now. 
I agree that we need to move to a better requirements system, and tbh auto_rx should really be run from within a virtualenv. I guess I can still have the instructions have the users install most packages from the system package manager, and the last few (flask-socketio, and soon simplekml) via a requirements.txt, which will let me pin versions.

darksidelemm, 2020-12-16:
Re-opening this until we have a requirements.txt and updated instructions sorted out.

darksidelemm, 2020-12-28:
This issue is mostly resolved in 1.4.0, however we should still look into updating to the latest version of socketio, will leave this open but re-name it

closed 2021-04-09 by darksidelemm