projecthorus/radiosonde_auto_rx #329: DFM Frame Number Wrapping

darksidelemm, 2020-12-16:
Currently the DFM frame number is taken from a single byte value in the telemetry, so it wraps at 255.

This has recently been seen to cause some issues with the tracker, which is expecting a frame number which monotonically increments throughout the flight. Issues also pop up in other places, e.g. https://sondehub.org/card/DFM-19048520  which was incorrectly sorting by frame number when it should be sorting by time.

I suggest that the DFM frame number instead be generated from the UTC timestamp, converted to a unix time in seconds. This is already the approach used for the M10 sondes, which don't have a frame number.

@rs1729

darksidelemm, 2020-12-16:
Also worth noting that this will be a breaking change. We will need to block DFM uploads to sondehub from stations older than 1.3.3 when this change is released.

rs1729, 2020-12-16:
For JSON output I could calculate gps week and tow from UTC date/time (neglecting leap seconds), and then the json frame number same way as for M10.

darksidelemm, 2020-12-16:
I think that would be more reliable - as long as all stations are consistent it will work fine, and on SondeHub we have the ability to block older station versions from uploading.

rs1729, 2020-12-16:
You can test
https://github.com/rs1729/RS/blob/test/demod/mod/dfm09mod.c
The JSON frame counter is now seconds since GPS 1980-01-06 (like M10 JSON counter).
However the DFM data is UTC, so leap seconds are not considered when calculating back. I don't know what the DFM will transmit when the next leap second is inserted. (If the time output would be 23h59m60s, the json-output would be skipped.)

darksidelemm, 2020-12-28:
Closed in https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.4.0

closed 2020-12-28 by darksidelemm