projecthorus/radiosonde_auto_rx #334: Ver Number on Web not updated on update Ver 1.3.2 to 1.3.3

Mbarelli, 2020-12-19:
The Version Number is not updated on web interface or I'm doing something wrong with updating to ver 1.3.3

TheSkorm, 2020-12-19:
The version for the web is auto populated. I don't think version 1.3.3 has been released yet so it will only work if your on the testing branch
`git branch testing; git pull`

https://github.com/projecthorus/radiosonde_auto_rx/blob/testing/auto_rx/autorx/__init__.py
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/web.py#L71
https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/templates/index.html#L61

darksidelemm, 2020-12-19:
Yes, 1.3.3 is still in beta testing. I would suggest not switching to it yet, as it requires a number of other changes to be made to function. If you switch to it now you will likely break your station.

closed 2020-12-19 by darksidelemm