projecthorus/radiosonde_auto_rx #342: Incorrect Pressure Readings

ZL1LAC, 2020-12-24:
Getting incorrect readings, the radiosonde is an RS41.  Not sure what is going on.

`Clb=-15.2m/s t=7.9C h=80.4% p=-1.0hPa 403.001 MHz Type=RS41-SG Radiosonde auto_rx v1.4.0`

-1.0hPa reading on all the data listing.

TheSkorm, 2020-12-24:
RS41-SG doesn't have a pressure sensor

ZL1LAC, 2020-12-24:
Well that's interesting I spoke to a person that launches them and he says
they do.

TheSkorm, 2020-12-24:
RS41-SGP has a pressure sensor, RS41-SG does not.

darksidelemm, 2020-12-24:
Both sondes I can see from NZ today were RS41-SG sondes, which are missing the pressure sensor. Some met bureaus (Australia included) decided to drop the pressure sensor and use GPS altitude instead. I believe the Vaisala ground station software will include a pressure reading, but it's based off GPS altitude, using an atmospheric density model.

closed 2020-12-24 by darksidelemm