projecthorus/radiosonde_auto_rx #343: update from 1.3.2 to 1.4.0

z3r0l1nk, 2020-12-24:
Hello,
Update was successful but after starting auto_rx I have the following error:
```
radiosonde auto_rx[229]: 2020-12-24 10:41:39,540 ERROR:Habitat - Unable to fetch UUIDs, retrying in 10 seconds - HTTPSConnectionPool(host='habitat.sondehub.org', port=443): Max retries exceeded with url: /_uuids?count=10 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7fae51d2e8>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
radiosonde auto_rx[229]: 2020-12-24 10:41:39,542 ERROR:Version - Error determining latest master version - HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /projecthorus/radiosonde_auto_rx/master/auto_rx/autorx/__init__.py (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object
at 0x7fae51d828>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')) 
```
But, it still works.

darksidelemm, 2020-12-24:
The "Temporary failure in name resolution" error indicates its an issue with your internet connection, unable to perform a DNS lookup. Nothing particular to auto_rx.

z3r0l1nk, 2020-12-24:
Hi,
Yes, it is from my end, because I have two active ISP's.
Sorry for the opened issue.

Kind regards,
Mihai

closed 2020-12-24 by z3r0l1nk