projecthorus/radiosonde_auto_rx #348: No Humidity nor pressure data from DFM09

MichelGreijmans, 2020-12-28:
I've managed to successfully track balloons launched from Curaçao, our local meteorology service uses the DFM09 devices/format. However I've never managed to decode any humidity nor pressure data sent out from these devices. Is this a compatibility/limitation with the DFS09 devices?

darksidelemm, 2020-12-28:
Decoding and calculating sensor data (temperature, pressure, humidity) requires a lot of knowledge of the inner workings of a radiosonde. Temperature is usually the easiest of the three, but humidity in particular requires a lot of calculations and correction data.
Remember that all of the telemetry from these sondes is reverse engineered (by @rs1729 ) based on received telemetry, and sometimes some 'truth' data from a ground-station. If you can provide a capture of an entire flight, along with the truth data from the associated ground station, that would help progress such efforts.
However, since the DFM09 is going out of production, I'm not sure it would be worth the effort. Better to focus on the incoming DFM17 sonde.

bazjo, 2020-12-28:
DFM-09 and DFM-17 really are not too far away regarding all things PTU at this point.

However, unlike RS41, DFMs are not transmitting calibration values and their telemetry has a lot more room for variation between types/customers, so determining the PTU data is not so straightforward compared to RS41

even in the best case, the ptu values obtained purely from telemetry will be off by such a huge amount that they could barely be used in a metrological setting. Met offices sync the calibration data into GrawMet via a local usb/uart connection from the sondes eeprom prior to flight.

darksidelemm, 2020-12-28:
Ouch... if it doesn't send the calibration data in the telemetry then that's a huge issue. Kudos on getting an even vaguely valid temperature readout from them!

rs1729, 2020-12-28:
Yes, that's a bit of a problem.
The temperature is close enough using the approximation of the NTC-Thermistor EPCOS B57540G0502 sensor. Although for low temperatures the approximation used might be a little off.
The raw humidity data is also clearly visible in the data, but without calibration I don't know how much it varies.
I think
https://www.imk-tro.kit.edu/4882.php
also mentions the manufacturer for the (DFM-06) humidity sensor: E+E HC103M2.
Most of the work was figuring out the floating point format in the data.

MichelGreijmans, 2020-12-28:
Would it be helpful if I ware to get you the official data from our local weather service? As I understand it the device is calibrated manually before launch.

rs1729, 2020-12-28:
Maybe this could help to reconstruct the calibration data after the flight for a particular radiosonde, but for decoding PTU of the radiosonde while in the air, you would need the calibration before/during the flight. The GrawMet-Software knows the calibration before launch, I guess, since it communicates with the radiosonde before launch, I guess.

drid, 2022-03-01:
Find attached temperature and humidity data as well as audio file of DFM09
Data exported as received by grawmet and graw ground station
Hope this helps
Pressure sensor is a BMP180 so calibration is done in BOSCH factory AFAIK

[202202181518076955_H.csv](https://github.com/projecthorus/radiosonde_auto_rx/files/8163925/202202181518076955_H.csv)
[202202181518076955_T.csv](https://github.com/projecthorus/radiosonde_auto_rx/files/8163926/202202181518076955_T.csv)
[gqrx_20220218_154652_402010000.zip](https://github.com/projecthorus/radiosonde_auto_rx/files/8163952/gqrx_20220218_154652_402010000.zip)

rs1729, 2022-04-29:
@einergehtnochrein
This is another example of Mode 4 and XDATA all zero.
https://github.com/rs1729/RS/issues/44

darksidelemm, 2024-11-16:
DFM09 not a priority anymore. There may still be some applicability for DFM17, but that can be a new issue if necessary.

closed 2024-11-16 by darksidelemm