projecthorus/radiosonde_auto_rx #352: Attempting to reset: /dev/bus/usb/001/013

Crille-ONCR16, 2021-01-03:
Hi have 3 sdr's on auto rx ... sometimes 1 sdr stop working. 
Attempting to reset: /dev/bus/usb/001/013
What cause this issue?

Crille-ONCR16, 2021-01-03:
Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:10,378 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:13,025 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:15,667 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:18,303 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:20,541 ERROR:RTLSDR - RTLSDR with serial #00000003 was not recovered after 5 reset attempts.
2021-01-03 12:42:20,542 ERROR:Scanner #00000003 - Unrecoverable RTLSDR error. Closing scan thread.
2021-01-03 12:42:20,543 INFO:Scanner #00000003 - Scanner Thread Closed.
2021-01-03 12:42:20,711 INFO:Task Manager - SDR #00000003 has been allocated to Scanner.
2021-01-03 12:42:21,299 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:23,911 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:26,522 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013
2021-01-03 12:42:29,105 INFO:RTLSDR - Attempting to reset: /dev/bus/usb/001/013

darksidelemm, 2021-01-03:
Hmm, after that many resets the SDR should actually be locked out from auto_rx. Is this what is happening?

There are known issues with running multiple RTLSDRs (in particular 3 or more) on Raspberry Pis. I only know of one station that has reliably had 3 RTLSDRs running, and he is using an external powered USB hub. It appears to be some limitation of the RPis USB bus, but there's also probably RTLSDR driver issues at play too...

darksidelemm, 2021-01-13:
Any further updates on this @Crille-ONCR16 ?

omegat, 2021-03-11:
... i had the same Problem today when I hooked up a second RTLSDR (nooelec v4) via a powered USB hub to my RPi3. After some rebooting, plugging in several configurations and testing (I was able to run rtl_power on both sticks simultaneously, so no bottleneck there) I stopped auto_rx.service and ran the python script via ssh. The script ran with no problems. After that everything (incl. the service) ran fine, even after reboot... 
So maybe some internal config problem?

BTW an cfg-option to allocate the logging path would be nice.

closed 2021-04-18 by darksidelemm