projecthorus/radiosonde_auto_rx #356: Radiosondy.info (SQ6KXY Radiosonde Tracker) Integration

kikosgc, 2021-01-08:
Hey guys,
I was wondering if it would be possible to forward (send), like we do with SondeHub + APRS-IS, the decoded received data to the radiosondy.info website (SQ6KXY Radiosonde Tracker).
There are a lot of people using multiple systems at the same time to receive and send the exact same data but to different services.  

Thanks and good chasing!

73 de F4VSE

darksidelemm, 2021-01-08:
Go take a closer look at the default aprs server in the example configuration file: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L162

We upload to radiosondy.info by default...

closed 2021-01-08 by darksidelemm

kikosgc, 2021-01-08:
My bad, Mark! Sorry about that... 😅😅