projecthorus/radiosonde_auto_rx #361: GPSD problem

sp3rat, 2021-01-15:
Hello Mark.
Sorry for my bad English.
I have a problem with auto_rx and gpsd on my Rasberry PI4.
I forward GPS Data over UDP from my mobile phone. The Rasberry Pi gpsd reads my forwarded Data (eg. gpsmon, cgps), but auto_rx does not read this data.
I don't know how to set gpsd in auto_rx or in chasemapper.
I am using the android "gpsd client" on my mobile phone to forward GPS Data.
Thank you for your help.
Jurek

darksidelemm, 2021-01-15:
It's very possible there are bugs in the gpsd implementation, I don't use it much.

Assuming your GPSD instance is on localhost:2947, can you try running (from within radiosonde_auto_rx/auto_rx):
python3 -m autorx.gpsd

This should just emit position data for about 30 seconds. Report any errors you see here.

kikosgc, 2021-01-29:
I tried it myself, couldn't get it working either. 
```pi@RPi-F4VSE:~/radiosonde_auto_rx/auto_rx $ python3 -m autorx.gpsd
2021-01-29 19:44:25,695 ERROR:GPSD - GPSDSocket connect exception: [Errno 111] Connection refused
2021-01-29 19:44:25,695 ERROR:GPSD - Connect failed. Waiting 10 seconds before re-trying.
2021-01-29 19:44:35,706 ERROR:GPSD - GPSDSocket connect exception: [Errno 111] Connection refused
2021-01-29 19:44:35,707 ERROR:GPSD - Connect failed. Waiting 10 seconds before re-trying.
```

xssfox, 2021-01-29:
This is almost certainly a gpsd configuration issue. Note that if your using systemd then gpsd connections are configured from ` /lib/systemd/system/gpsd.socket` 

You might want to check what `ListenStream=` is set to, it may be listening only on ipv4 or ipv6. You can check gpsd logs with `sudo service gpsd `

kikosgc, 2021-01-30:
> This is almost certainly a gpsd configuration issue. Note that if your using systemd then gpsd connections are configured from ` /lib/systemd/system/gpsd.socket`
>
> You might want to check what [...]

Yes for sure that is a GPSd problem in my case. I did eventually fix the problem above. But still not succeeded sent GPS data from  the Android phone (using  GPSd Client / Android-GPSd-Forwarder) to the GPSd server.
I'm going to try some other alternatives, but i'm convinced that the reported problem is with GPSd and not with radiosonde_auto_rx.

kikosgc, 2021-01-30:
Update: Everything working properly in my side.
```
pi@RPi-F4VSE:~ $ cd radiosonde_auto_rx/auto_rx
pi@RPi-F4VSE:~/radiosonde_auto_rx/auto_rx $ python3 -m autorx.gpsd
2021-01-30 10:31:40,512 INFO:GPSD - Connected to GPSD instance at 127.0.0.1
{'type': 'GPS', 'latitude': 51.003433933, 'longitude': 2.354985683, 'altitude': 26.0, 'speed': 0.0, 'valid': True} 

```

closed 2021-03-22 by darksidelemm