projecthorus/radiosonde_auto_rx #364: RS92 - GPS Ephemeris Downloading via FTP Broken

darksidelemm, 2021-01-24:
It looks like CDDIS closed off access via anonymous FTP in October 2020 (refer https://cddis.gsfc.nasa.gov/ ). Access is now only via HTTPS or ftp-ssl, both of which require a login (freely available, but would need to be added to the config somewhere).

This issue only affects decoding of RS92 sondes, which are obsolete and are hopefully on their way to being phased out.

Almanac downloading still works, though will result in a less accurate position solution.

mjgardes, 2021-01-24:
I got around that by changing to the esa server and update the file extensions.  I wanted to improve on that bit of code, but haven't got around to it yet.

The NWS offices near me are still launching RS92-NGPs and I don't see any indication that they or the other 18 or so NWS offices using the NGPs are looking into new models yet.  The US is pretty slow to adopt anything new.
```

  def get_ephemeris(destination="ephemeris.dat"):
    """ Download the latest GPS ephemeris file from the CDDIS's FTP server """
    try:
        logging.debug("GPS Grabber - Connecting to GSFC FTP Server...")
        ftp = ftplib.FTP("gssc.esa.int", timeout=10)
        ftp.login("anonymous", "anonymous")
        ftp.cwd("gnss/data/daily/%s/brdc/" % datetime.datetime.utcnow().strftime("%Y"))
        file_list = ftp.nlst()

        # We expect the latest files to be the last in the list.
        download_file = None
        file_suffix = datetime.datetime.utcnow().strftime("%yn.gz")

        if file_suffix in file_list[-1]:
            download_file = file_list[-1]
        elif file_suffix in file_list[-2]:
            download_file = file_list[-2]
        elif file_suffix in file_list[-3]:
            download_file = file_list[-3]
        elif file_suffix in file_list[-4]:
            download_file = file_list[-4]
        elif file_suffix in file_list[-5]:
            download_file = file_list[-5]
        else:
            logging.error("GPS Grabber - Could not find appropriate ephemeris file.")
            return None

        logging.debug(
            "GPS Grabber - Downloading ephemeris data file: %s" % download_file
        )

        # Download file.
        f_eph = open(destination + ".gz", "wb")
        ftp.retrbinary("RETR %s" % download_file, f_eph.write)
        f_eph.close()
        ftp.close()

        # Unzip file.
        os.system("gunzip -q -f ./%s" % (destination + ".gz"))

        logging.info(
            "GPS Grabber - Ephemeris downloaded to %s successfuly!" % destination
        )

        return destination
    except Exception as e:
        logging.error("GPS Grabber - Could not download ephemeris file. - %s" % str(e))
        return None
```

darksidelemm, 2021-01-25:
If you have a working solution, could you maybe submit a PR to the testing branch with it?

closed 2021-04-09 by darksidelemm