projecthorus/radiosonde_auto_rx #373: Altitude-based APRS-IS upload rate

Luceoria, 2021-01-28:
Current maximum APRS-IS upload rate is every 30 seconds. For high altitudes this is more than enough, but for lower altitudes it might not be an optimal solution. Especially if we want to know sonde's last position as close as possible to the ground, or when using a mobile decoder for sonde hunting.

My suggestion is as follows:

- Upload according to user settings, i.e. once a minute
- if altitude is less than N, upload the position in a shorter periods of time, i.e. 10s.

This seems like a more optimal solution and could also be a measure to prevent spamming APRS-IS servers with unneeded packets during flight on high altitudes.

darksidelemm, 2021-01-28:
It's worth noting that positions are uploaded every 30 seconds, but it's not just a 'snapshot' of the position at that instant.
A queue of packets are stored for each sonde, and at 30 second intervals the queue is dumped, and the most recent packet is uploaded. This means that the last received packet for each sonde should always end up being uploaded. 

Also, if you are chasing a sonde, you don't need to be using APRS-IS at all. You can be using better, local solutions like chasemapper which operate at a much higher update rate. 

Your solution would also result in a sonde that lands within range of a receive station continually being uploaded at a high rate.

Luceoria, 2021-01-29:
> This means that the last received packet for each sonde should always end up being uploaded.
Didn't know about that behavior. But that means some of the stations can sometimes send delayed packets from the past?

> Your solution would also result in a sonde that lands within range of a receive station continually being uploaded at a high rate.
Agree, but this could be easily resolved in the code, as altitude is not changing when sonde is on the ground.

As an example, this option exists in dxlAPRS and works just fine.

darksidelemm, 2021-01-29:
I upload the position objects using the HHMMSS timestamp format. APRS mapping system should make use of that timestamp and order track points appropriately. 
Note that this is really only going to occur if a station loses tracking during the beginning of the 30 seconds 'data collection window', and the oldest an uploaded position can be is that window time - 30 seconds by default.

I'm sure the situation of a landed radiosonde could be resolved in the code. The particular function that would need to be modified is this one: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/aprs.py#L687

However.. Given the last observed position should always make its way into APRS-IS, I don't really see this as a problem that needs solving?

jtmnt, 2021-03-02:
I have actually made this kind of change for myself and tested for many months. The reason for sending lower altitude packets more frequently is that when the sonde changes direction in lower altitude, then the prediction map showing the predicted landing spot on radiosondy.info will be more accurate. This has helped to find many sondes, days after they stopped transmitting. Only problem here could be the altitude of terrain in local area - since the altitude received is from above sea level.

You can see the change I made (more like a hack, not production redy code - overrides the configuration)

https://github.com/jtmnt/radiosonde_auto_rx/commit/b63750ed2c6fd6566646995209a0da063d4a4b1e

darksidelemm, 2021-04-18:
I think i'm going to drop this one unless a suitable PR is provided. 

With the new Sondehub v2 DB uploader we now upload *all* received telemetry into the Sondehub database, for display on the tracker. This resolve the issue of missing landing positions entirely. The Sondehub DB has an [open API](https://github.com/projecthorus/sondehub-infra/wiki/API-(Beta)), so if other radiosonde tracking websites want to access the data, they can. It's time to move on from thrashing APRS-IS and use something that is actually better suited to the task.

closed 2024-01-24 by darksidelemm