projecthorus/radiosonde_auto_rx #377: Email notification broken

macca100, 2021-02-01:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

When running the test email notification script it would fail with a missing dateutil module.  I had confirmed that 'python3-dateutil' was installed using apt as per the instructions. To resolve the issue I installed the module using PIP.

sudo pip install python-dateutil

This solved the problem.

darksidelemm, 2021-02-01:
The fact the issue was resolved on Raspbian by you running `pip install ...` (instead of `pip3`) suggests that you are running Python 2, in which case the package not being available makes sense.

I'd suggest making sure you are running auto_rx using Python 3, as I will no longer be supporting Python 2.

macca100, 2021-02-01:
Hi,
Yes, checking the version it came back as Python 2.7.16, however this PI4 was only recently built using the latest 32 bit image from the imaging tool. 

After doing some reading It looks like the image comes with Python 2 by default and we install Python3. 

I used the command found on the wiki to test the emails, I didn't spot the difference in the python it was using. Tested using python3 and it ran fine.

https://github.com/projecthorus/radiosonde_auto_rx/wiki/Configuration-Settings

$ cd ~/radiosonde_auto_rx/auto_rx/
$ python -m autorx.email_notification

Is it possible for the wiki to be updated for python3

darksidelemm, 2021-02-01:
Ahh yep, all updated now, thanks!

closed 2021-02-01 by darksidelemm