projecthorus/radiosonde_auto_rx #380: No Velocity Data for IMet?

ZigiWalter, 2021-02-03:
Hi,

It seems that the log files of IMet sondes always specify the vertical and horizontal velocities as zero.
I guess it's not an auto_rx issue, but rather because speed data is not transmitted by the sonde or not recovered by the decoder, right?

Thanks.

darksidelemm, 2021-02-03:
Probably the case.

The 'spec' for the imet telemetry ( https://github.com/rs1729/RS/blob/master/imet/imet1rs-binary.pdf ) does talk about an enhanced GPS packet with this information, however I'm not sure all radiosondes send this.
@rs1729  do you know if that enhanced GPS packet is commonly sent?

rs1729, 2021-02-03:
No, so far I have only seen the packets 01 02 (GPS) and 01 04 (ePTU).
I guess it would send either GPS (01 02) or eGPS (01 05).
So if there is no GPS position from the decoder, this could mean there is eGPS instead of GPS transmitted.
If someone has such a sample, I would implement it.

ZigiWalter, 2021-02-03:
Thanks for the quick answers!
Let me know how I can provide a sample, or somehow check by myself whether it's GPS or eGPS.
If it helps - I think they are all iMet-4C.

Thanks.

rs1729, 2021-02-03:
I would think that the imet-4 would either send GPS packets or eGPS, since eGPS includes the GPS data.
Right now the decoder looks only for GPS and ePTU packets. (All iMet-4 and iMet-1-RS I know send these two packets.)
So if your iMet-4C would have eGPS (01 05) instead of GPS (01 02), the decoder would not recognize it (at least I don't remember implementing eGPS) and there would be no GPS position. If you have a good signal and no position output, you can make a short recording and I can look at it.

Here I describe a simple method how to get the raw data on linux:
https://www.fingers-welt.de/phpBB/viewtopic.php?f=14&t=43&start=1000#p88845
(german)
For this you need a FM (mono) audio recording and `minimodem`, then
```
minimodem --rx 1200 -f imet4fm_mono.wav > imet1rsb-bell202.bytes
```
Usually it shows frames like
```
01 02 .... 01 04 ....
```

rs1729, 2021-02-04:
Velocity would be nice for tracking, and most radiosondes transmit velocity data.
For meteorology I hear it is not so important. You have to filter out the pendulum motion anyway, the positions give the same results I guess.

TheSkorm, 2021-02-04:
It's certainly a problem!
[image]

ZigiWalter, 2021-02-24:
@rs1729, @darksidelemm  - thanks!
I've recorded a couple of IMET sondes (using gqrx), converted to Mono and ran minimodem according to the instructions (I think). However, I could only get the kind of output pasted below. What am I doing wrong...?

Thanks again.

### NOCARRIER ndata=26 confidence=4.564 ampl=0.083 bps=1198.16 (0.2% slow) ###
### CARRIER 1200 @ 1200.0 Hz ###
d�B|�B�(

�56����F4((���
### NOCARRIER ndata=38 confidence=4.577 ampl=0.083 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
G�Bm�B�(

        f
7����F4::���
### NOCARRIER ndata=38 confidence=4.542 ampl=0.083 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
1�Bc�B�(

        ��8����P4II���
### NOCARRIER ndata=38 confidence=4.617 ampl=0.084 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
�Bb�B�(

rs1729, 2021-02-24:
Don't know, do you have the recording somewhere?
What was the exact command you used? Did you redirect the output into a file? Because the output is binary.

E.g.

> `minimodem --rx 1200 -f imet4fm_mono.wav > imet4-bell202.bytes`
> 
> `xxd -g 1 -c 38 imet4-bell202.bytes`

Or pipe it through xxd
```
minimodem --rx 1200 -f imet4fm_mono.wav | xxd -g 1 -c 38
```

ZigiWalter, 2021-02-24:
Here's the mono recording: http://www.filedropper.com/gqrx20210224175951404997000mono

I've tried decoding using this command-line:
minimodem --rx 1200 -f  gqrx_20210224_175951_404997000_mono.wav

From the above instructions, I didn't relaize I also need to redirect the output to xxd. If I do that, I get:

`00000000: 01 02 aa dd 03 42 7e e2 0e 42 a6 28 0c 12 00 08 e5 15 01 04 34 04 98 f8 14 4d 01 02 64 dd 03 42 7c e2 0e 42 b2 28  .....B~..B.(........4....M..d..B|..B.(
00000026: 0c 12 00 0a 9b 35 01 04 36 04 d2 c9 00 a2 f8 46 05 34 28 00 28 00 96 f8 e7 0f 01 02 47 dd 03 42 6d e2 0e 42 b6 28  .....5..6......F.4(.(.......G..Bm..B.(
0000004c: 0c 12 00 0b 66 0a 01 04 37 04 b4 c9 00 a0 f8 46 05 34 3a 00 3a 00 91 f8 ac 13 01 02 31 dd 03 42 63 e2 0e 42 bb 28  ....f...7......F.4:.:.......1..Bc..B.(
00000072: 0c 12 00 0c 1a ba 01 04 38 04 93 c9 00 9d f8 50 05 34 49 00 49 00 8d f8 f1 ba 01 02 18 dd 03 42 62 e2 0e 42 c0 28  ........8......P.4I.I..........Bb..B.(
00000098: 0c 12 00 0d ee 25 01 04 39 04 72 c9 00 99 f8 5a 05 34 53 00 53 00 8c f8 a1 95 01 02 03 dd 03 42 69 e2 0e 42 c5 28  .....%..9.r....Z.4S.S..........Bi..B.(
000000be: 0c 12 00 0e c6 fc 01 04 3a 04 4b c9 00 95 f8 64 05 34 56 00 56 00 88 f8 c5 d6 01 02 ec dc 03 42 74 e2 0e 42 cb 28  ........:.K....d.4V.V..........Bt..B.(
000000e4: 0c 12 00 0f 75 d3 01 04 3b 04 23 c9 00 90 f8 64 05 34 55 00 55 00 85 f8 9a 87 01 02 d5 dc 03 42 7e e2 0e 42 d1 28  ....u...;.#....d.4U.U..........B~..B.(
0000010a: 0c 12 00 10 55 78 01 04 3c 04 eb c8 00 8a f8 78 05 34 51 00 51 00 81 f8 91 ab 01 02 bb dc 03 42 8c e2 0e 42 da 28  ....Ux..<......x.4Q.Q..........B...B.(
00000130: 0c 12 00 11 35 b8 01 04 3d 04 b2 c8 00 84 f8 82 05 34 0b 00 4b 00 7d f8 55 60 01 02 96 dc 03 42 9b e2 0e 42 e1 28  ....5...=........4..K.}.U`.....B...B.(
00000156: 0c 12 00 12 91 f9 01 04 3e 04 86 c8 00 7d f8 a0 05 34 44 00 44 00 77 f8 cc f2 ff 01 02 6a dc 03 42 a7 e2 0e 42 e7  ........>....}...4D.D.w......j..B...B.
0000017c: 28 0c 12 00 13 77 eb 01 04 3f 04 57 c8 00 76 f8 be 05 34 3a 00 3a 00 74 f8 2f 3f 01 02 36 dc 03 42 b0 e2 0e 42 ec  (....w...?.W..v...4:.:.t./?..6..B...B.
000001a2: 28 0c 12 00 14 b7 44 01 04 40 04 20 c8 00 71 f8 e6 05 34 32 00 32 00 6f f8 e4 ff 01 02 04 5c 03 42 b7 e2 0e 42 f4  (.....D..@. ..q...42.2.o......\.B...B.
000001c8: 28 0c 12 00 15 01 a3 01 04 41 04 e2 c7 00 6b f8 0e 06 34 29 00 29 00 67 f8 a3 a0 ff ff 01 02 da db 03 42 bf e2 0e  (........A....k...4).).g..........B...
000001ee: 42 fd 28 0c 12 00 16 a3 5b 01 04 42 04 ac c7 00 64 f8 2c 06 34 20 00 20 00 61 f8 7b c5                             B.(.....[..B....d.,.4 . .a.{.
`

Thanks.

rs1729, 2021-02-24:
I guess minimodem is only to be used if nothing else is available.

If you use the decoder with the `-r` option, you get the raw bytes between idle if the first byte is 01.
https://github.com/rs1729/RS/blob/master/imet/imet1rs_dft.c
The packet description says that all the packets start with SOH=0x01.
https://github.com/rs1729/RS/blob/master/imet/imet1rs-binary.pdf
Only "01 02" and "01 04" are decoded, as I said, I have only seen these two.
However even if you see e.g. "01 03" or "01 05", it could be just a decoding error, so one would have to look at these packets and their CRC. If most of the frames are decoded as "01 02 ... 01 04", other packets probably just have errors.

darksidelemm, 2021-04-18:
Any updates on this? @ZigiWalter can you provide some more signal recordings?
Otherwise I would like to close this issue...

ZigiWalter, 2021-04-18:
Sure - should have new recordings ready in a day or two.

Thanks.

ZigiWalter, 2021-04-18:
Here's another recording:
http://www.filedropper.com/gqrx20210418181217401997800

I ran:
```
>sox ~/gqrx_20210418_181217_401997800.wav ~/gqrx_20210418_181217_401997800_mono.wav remix 1
>minimodem --rx 1200 -f ~/gqrx_20210418_181217_401997800_mono.wav | xxd -g 1 -c 38
```

and got:
```
### CARRIER 1200 @ 1200.0 Hz ###

### NOCARRIER ndata=13 confidence=5.038 ampl=0.087 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000000: ee 04 38 09 35 2f 0a 2f 0a ec 04 85 ae 01 02 23 15 03 42 0b b9 0d 42 d5 1e 0c 12 0c 24 f1 c7 01 04 20 04 00 1d 01  ..8.5/./.......#..B...B.....$.... ....

### NOCARRIER ndata=38 confidence=4.858 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000026: ec 04 56 09 35 2a 0a 2a 0a e7 04 74 a6 01 02 23 15 03 42 04 b9 0d 42 da 1e 0c 12 0c 25 72 54 01 04 21 04 e0 1c 01  ..V.5*.*...t...#..B...B.....%rT..!....

### NOCARRIER ndata=38 confidence=4.725 ampl=0.085 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###

### NOCARRIER ndata=2 confidence=1.728 ampl=0.097 bps=1176.47 (2.0% slow) ###
### CARRIER 1200 @ 1200.0 Hz ###

### NOCARRIER ndata=2 confidence=2.156 ampl=0.101 bps=1176.47 (2.0% slow) ###
### CARRIER 1200 @ 1200.0 Hz ###
0000004c: eb 04 60 09 35 28 0a 28 0a e3 04 55 34 ff ff bf fa 01 02 1e 15 03 42 ff b8 0d 42 dd 1e 0c 12 0c 26 97 f3 01 04 22  ..`.5(.(...U4.........B...B.....&...."

### NOCARRIER ndata=38 confidence=4.549 ampl=0.087 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000072: 04 be 1c 01 e9 04 6a 09 35 25 0a 25 0a e2 04 bb e2 01 02 1d 15 03 42 f9 b8 0d 42 e2 1e 0c 12 0c 27 93 41 01 04 23  ......j.5%.%..........B...B.....'.A..#

### NOCARRIER ndata=38 confidence=4.833 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000098: 04 9a 1c 01 e7 04 60 09 35 23 0a 23 0a e1 04 30 65 01 02 1e 15 03 42 e6 b8 0d 42 e5 1e 0c 12 0c 28 e5 6b 01 04 24  ......`.5#.#...0e.....B...B.....(.k..$

### NOCARRIER ndata=38 confidence=4.427 ampl=0.085 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
000000be: 04 6c 1c 01 e4 04 60 09 35 20 0a 20 0a dc 04 2a 6b 01 02 23 15 03 42 df b8 0d 42 eb 1e 0c 12 0c 29 cb eb 01 04 25  .l....`.5 . ...*k..#..B...B.....)....%

### NOCARRIER ndata=38 confidence=4.616 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
000000e4: 04 42 1c 01 e2 04 60 09 35 1d 0a 1d 0a db 04 e3 4f 01 02 24 15 03 42 e1 b8 0d 42 ef 1e 0c 12 0c 2a 19 40 01 04 26  .B....`.5.......O..$..B...B.....*.@..&

### NOCARRIER ndata=38 confidence=4.949 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
0000010a: 04 1f 1c 01 dd 04 6a 09 35 1a 0a 1a 0a d8 04 dc 36 01 02 20 15 03 42 d2 b8 0d 42 f2 1e 0c 12 0c 2b bc 57 01 04 27  ......j.5.......6.. ..B...B.....+.W..'

### NOCARRIER ndata=38 confidence=4.865 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000130: 04 fb 1b 01 d9 04 6a 09 35 18 0a 18 0a d5 04 e3 82 01 02 28 15 03 42 c7 b8 0d 42 f6 1e 0c 12 0c 2c 77 9c 01 04 28  ......j.5..........(..B...B.....,w...(

### NOCARRIER ndata=38 confidence=5.115 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
00000156: 04 d8 1b 01 d6 04 74 09 35 16 0a 16 0a d1 04 cf 73 01 02 2d 15 03 42 c5 b8 0d 42 f9 1e 0c 12 0c 2d fa f0 01 04 29  ......t.5.......s..-..B...B.....-....)

### NOCARRIER ndata=38 confidence=4.984 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###
0000017c: 04 b2 1b 01 d1 04 74 09 35 13 0a 13 0a ce 04 09 3a 01 02 2e 15 03 42 bc b8 0d 42 fe 1e 0c 12 0c 2e 75 fe 01 04 2a  ......t.5.......:.....B...B......u...*

### NOCARRIER ndata=38 confidence=4.902 ampl=0.086 bps=1200.00 (rate perfect) ###
### CARRIER 1200 @ 1200.0 Hz ###

### NOCARRIER ndata=6 confidence=4.649 ampl=0.084 bps=1200.00 (rate perfect) ###
000001a2: 04 92 1b 01 cb 04 74 09 35 11 0a 11 0a cc 04 2b f2 01 02 2c 15 03 42                                               ......t.5......+...,..B
```

Thanks again.

rs1729, 2021-04-18:
`./imet1rs_dft -r gqrx_20210418_181217_401997800.wav`
shows only `01 02` and `01 04` packets:
```
01 02 23 15 03 42 0B B9 0D 42 D5 1E 0C 12 0C 24 F1 C7 01 04 20 04 00 1D 01 EC 04 56 09 35 2A 0A 2A 0A E7 04 74 A6 
(18:12:36)  lat: 32.770641°  lon: 35.430706°  alt: 2893m  sats: 12  #  CRC: F1C7 - F1C7 [OK]
[1056]  P:729.60mb  T:12.60°C  U:23.90%  bat:5.3V  #  CRC: 74A6 - 74A6 [OK]

01 02 23 15 03 42 04 B9 0D 42 DA 1E 0C 12 0C 25 72 54 01 04 21 04 E0 1C 01 EB 04 60 09 35 28 0A 28 0A E3 04 55 34 
(18:12:37)  lat: 32.770641°  lon: 35.430679°  alt: 2898m  sats: 12  #  CRC: 7254 - 7254 [OK]
[1057]  P:729.28mb  T:12.59°C  U:24.00%  bat:5.3V  #  CRC: 5534 - 5534 [OK]

01 02 1E 15 03 42 FF B8 0D 42 DD 1E 0C 12 0C 26 97 F3 01 04 22 04 BE 1C 01 E9 04 6A 09 35 25 0A 25 0A E2 04 BB E2 
(18:12:38)  lat: 32.770622°  lon: 35.430660°  alt: 2901m  sats: 12  #  CRC: 97F3 - 97F3 [OK]
[1058]  P:728.94mb  T:12.57°C  U:24.10%  bat:5.3V  #  CRC: BBE2 - BBE2 [OK]

01 02 1D 15 03 42 F9 B8 0D 42 E2 1E 0C 12 0C 27 93 41 01 04 23 04 9A 1C 01 E7 04 60 09 35 23 0A 23 0A E1 04 30 65 
(18:12:39)  lat: 32.770618°  lon: 35.430637°  alt: 2906m  sats: 12  #  CRC: 9341 - 9341 [OK]
[1059]  P:728.58mb  T:12.55°C  U:24.00%  bat:5.3V  #  CRC: 3065 - 3065 [OK]

01 02 1E 15 03 42 E6 B8 0D 42 E5 1E 0C 12 0C 28 E5 6B 01 04 24 04 6C 1C 01 E4 04 60 09 35 20 0A 20 0A DC 04 2A 6B 
(18:12:40)  lat: 32.770622°  lon: 35.430565°  alt: 2909m  sats: 12  #  CRC: E56B - E56B [OK]
[1060]  P:728.12mb  T:12.52°C  U:24.00%  bat:5.3V  #  CRC: 2A6B - 2A6B [OK]

01 02 23 15 03 42 DF B8 0D 42 EB 1E 0C 12 0C 29 CB EB 01 04 25 04 42 1C 01 E2 04 60 09 35 1D 0A 1D 0A DB 04 E3 4F 
(18:12:41)  lat: 32.770641°  lon: 35.430538°  alt: 2915m  sats: 12  #  CRC: CBEB - CBEB [OK]
[1061]  P:727.70mb  T:12.50°C  U:24.00%  bat:5.3V  #  CRC: E34F - E34F [OK]

01 02 24 15 03 42 E1 B8 0D 42 EF 1E 0C 12 0C 2A 19 40 01 04 26 04 1F 1C 01 DD 04 6A 09 35 1A 0A 1A 0A D8 04 DC 36 
(18:12:42)  lat: 32.770645°  lon: 35.430546°  alt: 2919m  sats: 12  #  CRC: 1940 - 1940 [OK]
[1062]  P:727.35mb  T:12.45°C  U:24.10%  bat:5.3V  #  CRC: DC36 - DC36 [OK]

01 02 20 15 03 42 D2 B8 0D 42 F2 1E 0C 12 0C 2B BC 57 01 04 27 04 FB 1B 01 D9 04 6A 09 35 18 0A 18 0A D5 04 E3 82 
(18:12:43)  lat: 32.770630°  lon: 35.430489°  alt: 2922m  sats: 12  #  CRC: BC57 - BC57 [OK]
[1063]  P:726.99mb  T:12.41°C  U:24.10%  bat:5.3V  #  CRC: E382 - E382 [OK]

01 02 28 15 03 42 C7 B8 0D 42 F6 1E 0C 12 0C 2C 77 9C 01 04 28 04 D8 1B 01 D6 04 74 09 35 16 0A 16 0A D1 04 CF 73 
(18:12:44)  lat: 32.770660°  lon: 35.430447°  alt: 2926m  sats: 12  #  CRC: 779C - 779C [OK]
[1064]  P:726.64mb  T:12.38°C  U:24.20%  bat:5.3V  #  CRC: CF73 - CF73 [OK]

01 02 2D 15 03 42 C5 B8 0D 42 F9 1E 0C 12 0C 2D FA F0 01 04 29 04 B2 1B 01 D1 04 74 09 35 13 0A 13 0A CE 04 09 3A 
(18:12:45)  lat: 32.770679°  lon: 35.430439°  alt: 2929m  sats: 12  #  CRC: FAF0 - FAF0 [OK]
[1065]  P:726.26mb  T:12.33°C  U:24.20%  bat:5.3V  #  CRC: 093A - 093A [OK]

01 02 2E 15 03 42 BC B8 0D 42 FE 1E 0C 12 0C 2E 75 FE 01 04 2A 04 92 1B 01 CB 04 74 09 35 11 0A 11 0A CC 04 2B F2 
(18:12:46)  lat: 32.770683°  lon: 35.430405°  alt: 2934m  sats: 12  #  CRC: 75FE - 75FE [OK]
[1066]  P:725.94mb  T:12.27°C  U:24.20%  bat:5.3V  #  CRC: 2BF2 - 2BF2 [OK]
```

closed 2021-08-15 by darksidelemm