projecthorus/radiosonde_auto_rx #383: Add the ability to have imperial units displayed on the web interface

Reid-n0rc, 2021-02-07:
It would be nice to have the ability to display units as imperial here in the states.

I can start working on this or someone that knows that they're doing can do it too.

darksidelemm, 2021-02-07:
Not sure what you mean here?
I'm not sure i'd want to support an automatic switching to imperial units based on location. I'm not sure *everybody* in US would want to be using imperial measurements for chasing, so keeping it as a config option is a better solution.

Reid-n0rc, 2021-02-07:
No, be able to define if metric or imperial units are used from the config file

darksidelemm, 2021-02-07:
Oh sorry, I just realised this was on the auto_rx repository, not the chasemapper one :-)

I guess this is a maybe, if someone wants to have a go at implementing it? It's not really something I want to work on.

Given that ... every... radiosonde emits data in metric units, I don't see much value in converting things like temperature to imperial units, but maybe altitude or range?

Reid-n0rc, 2021-02-07:
I'll give it a try.

darksidelemm, 2021-02-07:
If you're going to do it, please target the testing branch. You will need to:
- Add another config option (preferably in the 'web') section of the config file.
- Add the default setting to the defaults here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/config.py#L55
- Parse this configuration file option handling the case where it has not been added. Default to metric. Example of where to do this here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/config.py#L489
- Change the web interface to change units based on the configuration file option, which will be accessible in the autorx_config object after page load: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/templates/index.html#L52
- Change the table column names based on the user units selection: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/templates/index.html#L172
- Convert units as they are displayed, probably in this function: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/templates/index.html#L199

Reid-n0rc, 2021-02-07:
Thanks Mark that's a huge help.

g7gpr, 2021-03-29:
Given that this has now been implemented in chasemapper, can I propose that this is closed?

darksidelemm, 2021-03-29:
Reid was specifically after it in auto_Rx as well..

However it's worth noting that the web interface has recently changed a lot, so the links in the post above will be invalid.

LukePrior, 2021-04-03:
This will be quite easy to implement if you want,

You will just need to define a few custom formaters for the columns which contain values that need to be converted. You will need to create a new tickbox in the settings menu to disable/enable it with it set to off by default. Then in the Tabulator formatted check if the option is set and then apply the conversion. You would also need to figure out how to change header name.

See here: http://tabulator.info/docs/4.9/format

LukePrior, 2021-04-04:
I've got this working locally, I just need to fix an issue with the table headers not automatically updating when the option is enabled.

closed 2021-04-09 by darksidelemm