projecthorus/radiosonde_auto_rx #385: Orange Pi installation

MCmakerPro, 2021-02-14:
Is it possible to optimize for orange pi boards? I installed on an orange pi zero, but the decoding process seems to stall the CPU only after a frame or two and it reverts back to scanning. There should be enough performance for it to work.

darksidelemm, 2021-02-14:
If an Orange Pi Zero is anything like a Pi Zero W then it's going to be on the edge. You might try the latest testing branch version, which enables some better compile-time optimisations for the decoders, but it might still not work.

I recommend something at least RPi 3 B+ performance or better... I don't see any point in spending time optimising for underpowered systems when it works on fairly easily available hardware.

closed 2021-02-14 by darksidelemm

MCmakerPro, 2021-02-14:
The orange pi zero should be a bit more powerful than the rpi zero as it has a quad-core 1200MHz cpu. But yes, I have had great success with a Pi 4. I will try the testing branch

darksidelemm, 2021-02-14:
I've had success running auto_rx on even a RPi 2 B... so if the orange pi has 4 cores then it should work fine.

Are you sure its a CPU usage issue and not something else? Even without optimisations the CPU usage of the decoders is fairly modest.