projecthorus/radiosonde_auto_rx #392: Add the ability to send notifications via telegram

superingole, 2021-02-22:
It would be nice the possibilty to send the notifications, which are actually send only via email, 
can be send via telegram messenger to groups or bots

I cant programming pyhton code, no knowledge in pyhton :-(

regards

darksidelemm, 2021-02-22:
I don't want to add support for something like this in the main auto_rx code... Next I'll have to support IRC, Discord, etc... 
The UDP broadcast output exists in part so these kind of features can be written by someone with the time and inclination to do so. An example bit of python code to listen for the UDP messages is here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/utils/horus_udp_receiver_example.py

The additional logic that would need to be written is:
- All the Telegram/Discord/whatever connection stuff.
- On first message for a particular sonde ID, send your message.
- When sonde ID is descending, then goes below some particular altitude, send the landing alert.

closed 2021-02-22 by darksidelemm

snh, 2021-02-23:
@superingole Running something like https://github.com/KostyaEsmukov/smtp_to_telegram alongside `radiosonde_auto_rx` may be worth investigating.

ch3p4ll3, 2024-12-18:
Hi, I created [this](https://github.com/ch3p4ll3/radiosonde_auto_rx_notifier) project that allows you to configure notifications to be sent if a radiosonde is below a threshold height and is within a range from coordinates. 

For the notifications I relied on [apprise](https://github.com/caronc/apprise) so you have a ton of services available with which you can send notifications. The project is still under development and can be used in a docker compose along with autorx