projecthorus/radiosonde_auto_rx #396: Decoder should not timeout when frequency and sonde type is provided in cmdline

Luceoria, 2021-02-27:
If auto_rx.py is called using command line with both sonde frequency and type provided, decoder process times out after rx_timeout and keeps on scanning/recognizing sonde type.
When frequency and type is provided by user, I believe there is no need to timeout the decoder.

Example command:
python3 auto_rx.py -f 403.000 -m MRZ

Currently this can be workarounded by setting rx_timeout in station.cfg to some high value.

I also tried the "-t" option, but it seems to timeout whole auto_rx process.

darksidelemm, 2021-02-27:
Hmm, yes this is probably the behaviour that should be implemented. Currently there is no timeout for the UDP mode, so it's probably possible to make it not timeout when a specific mode is provided. Will look into it

darksidelemm, 2021-02-28:
OK, this is now the default behaviour if a radiosonde model is manually specified. You will need to update to the latest testing branch commit (now beta13).

closed 2021-02-28 by darksidelemm