projecthorus/radiosonde_auto_rx #409: Viewing Past Radiosonde

LukePrior, 2021-03-22:
Hi,

I've got the program up and running and really like how I can view tracked items in real time and see where they have been. 

I know this information is all saved to the log files,   so I was wondering if it's possible to view these previous flights from the web interface or extend the time before they disappear?

Thanks

darksidelemm, 2021-03-22:
It's been thought about, but not really something I've had time to add to the web interface. I thought some others were working on it, but not sure where it's up to.

You can extend the history of the web interface with this setting: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L354

As for looking at seriously historical data, querying the various databases that auto_rx pushes data into (sondehub, radiosondy) is possibly a better option, as then you get data from more than just your station.

LukePrior, 2021-03-22:
No worries just thought I would ask.

Do you have any knowledge on what is a safe value to set the history to for the web interface, I only detect one launch a day which lasts for about 2 hours. I'm finding that I am getting rate limited more often than not with sondehub so I prefer to look at my own data. Is there any way to increase my chances of not getting rate limited due to high server load (time is at default 30s).

Cheers

darksidelemm, 2021-03-22:
You may be getting rate limited into Sondehub v1, but this is only rate limiting into the habitat database for display on tracker.sondehub.org Your data is still accessible directly from the sondehub db, via sondehub.org/card/<serial> 
However, this is all for Sondehub v1. Sondehub v2 ingests *all* telemetry that you receive, and will allow access to it via a few means, including a re-worked tracker interface (currently under development). This is coming in the next auto_rx release.

As for how much history you can show, it really depends on what your web browser can handle, but i've seen people running a few days.

closed 2021-03-22 by darksidelemm

fkemps, 2021-03-22:
Interesting, how to specify the Serial Number?

***  Please note that this message and any attachments may contain confidential and proprietary material and information and are intended only for the use of the intended recipient(s). If you are not the intended recipient, you are hereby notified that any review, use, disclosure, dissemination, distribution or copying of this message and any attachments is strictly prohibited. If you have received this email in error, please immediately notify the sender and destroy this e-mail and any attachments and all copies, whether electronic or printed. Please also note that any views, opinions, conclusions or commitments expressed in this message are those of the individual sender and do not necessarily reflect the views of Fortinet, Inc., its affiliates, and emails are not binding on Fortinet and only a writing manually signed by Fortinet's General Counsel can be a binding commitment of Fortinet to Fortinet's customers or partners. Thank you. ***

darksidelemm, 2021-03-22:
As an example: https://sondehub.org/card/S1130567

Use what you would see on tracker.sondehub.org but without the RS_ prefix.

Note that this website will change to point at the new Sondehub v2 DB at some point in the future.

LukePrior, 2021-03-22:
What I do think could be useful and easier to implement would be a simple drop down list at the bottom of the page which checks the saved logs and provides a line for each tracked Radiosonde with basic information like date, type, and flight time along with a link to view it on sondehub cards.

darksidelemm, 2021-03-22:
Sure... I think that's what was being developed by someone... but I can't remember who! Hopefully they pop up again... 
I have limited time and energy to work on this, and i'm mainly focused on adding core functionality, like new radiosonde types, or new data export functions (e.g. to Sondehub v2). I'd really like it if others would step up and help out with the UI side of things...