projecthorus/radiosonde_auto_rx #414: Interface Questions

LukePrior, 2021-03-25:
Hey,

I'm working on integrating my new interface with the existing functions to populate data, I've got logs, version, station location, current task all configured and working. I'm having a few issues getting the scan graph and telementary archive working, the scan graph is rendering put all the spectra points are too large see attached image.

[image]

What do I need to change to adjust the size and make it look more like the current one, I couldn't see anything obvious.

The telementary archive draw the line ok if I specify a colour, but I'm not sure where colour_values, sondeAscentIcons, sondeDescentIcons come from. I've highlighted them in the image below.

[image]

Cheers

darksidelemm, 2021-03-25:
Not sure why you raised this as an issue, its better to continue this discussion in the PR.

I wrote the interface so long ago so I don't remember much about these points, but looking at the code:

- Icons are created here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/static/js/utils.js#L12
- Scan chart created here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/static/js/scan_chart.js#L8 
- Main scan results are a line plot, not a scatter plot. Only the peak results are a scatter plot.

LukePrior, 2021-03-27:
Will continue in PR.

closed 2021-03-27 by LukePrior