projecthorus/radiosonde_auto_rx #424: Update socketio.js version so we can use latest flast-socketio version

darksidelemm, 2021-03-30:
Being pinned to an old version worries me a little.

Does the current latest socket.io.js version work with the latest flask-socketio?

LukePrior, 2021-03-31:
On the testing branch we just updated to 1.7.4 which was released May 7, 2017.

The latest version is 4.0.0 which was released 27 days ago.

The breaking changes for 1 --> 2, 2 --> 3, and 3 --> 4 can be found below.

https://github.com/socketio/socket.io/releases/tag/2.0.0
https://socket.io/docs/migrating-from-2-x-to-3-0/
https://socket.io/docs/v3/migrating-from-3-x-to-4-0/

I found this compatibility table on Flask-Socketio Documentation:

[image]

The table seems to suggest v3 is the latest supported, the latest Flask-Socketio release is from December 2020

darksidelemm, 2021-03-31:
Sounds good... I think we'll slate this change for the 1.5.1 release.

darksidelemm, 2021-03-31:
Well socket.js 4.0.0 seems to work OK with flask-socketio 5.0.1.

I'll merge this into the testing branch, and we can keep an eye out for any bugs...

LukePrior, 2021-03-31:
Sounds good

darksidelemm, 2021-03-31:
Hmm, one issue i did just encounter, trying to get flask-socketio updated on one of my remote stations, is that one of the flask-socketio dependencies (bidict), doesn't support Python 3.5 (which is the python version in Raspbian stretch). It needs Python 3.7 at a minimum. 

This means stations still on Raspbian stretch (like my remote station) will need to update to buster. I'm in the process of doing a dist-upgrade on my station. Hopefully it works, otherwise i'm going to need to go for a drive!

darksidelemm, 2021-03-31:
Welp, dist-upgrade failed. Time to build up a new RPi and go for a drive...

mhaberler, 2021-03-31:
> Welp, dist-upgrade failed. Time to build up a new RPi and go for a drive...

I had an rpi problem (multiple rtlsdr's) which would not reboot properly after a reboot command, but came up fine after powercycle

it turned out tweaking /boot/config.txt helped:
```
# try to fix reboot hang problems
# see https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&sid=9272af747ff28c1f9f35295e10bebc78
force_turbo=1
over_voltage=2
```

too late for this cycle though..

closed 2021-04-09 by darksidelemm