projecthorus/radiosonde_auto_rx #437: False Landing Notifications for IMet Sondes

ZigiWalter, 2021-04-05:
HI,

I've started to get incorrect landing notifications.

I suspect that sondes that are not sending their velocity are considered as having a vertical velocity of -9999.0 m/s (at least according to IMet log files). If so, the "descending_trip" counter in autorx/email_notification.py is increased any time such a sonde is received anywhere lower than "landing_altitude_threshold", regardless of its speed and direction.

As a result, landing notifications are sent upon launches, when the sonde is rising.

Is there something to do other than disabling landing notifications altogether (including for "normal" sondes)?
I assume that calculating the velocities in auto_rx itself is not something we want to do. If so, maybe something as simple as not sending notifications only for these kind of sondes, by adding a check for "-9999.0" before increasing the "descending_trip" counter? 

Thanks.

darksidelemm, 2021-04-05:
Hmm that's an interesting one. It hadn't twigged to me that the iMet's didn't send that information. I'll have a think about it. 

Just filtering out '-9999.0' is going to mean no landing notifications will be provided for sonde that don't provided vel_v data.

Calculating the ascent rate isn't that difficult to do, I already do that in chasemapper, so I could do that instead of using the data supplied in the telemetry.

darksidelemm, 2021-04-05:
As it turns out, I already have the code to do the ongoing ascent/descent rate calculations in auto_rx (GenericTrack within geometry.py), so I should be able to switch to using that to calculate the rates. I need to be a little bit careful to avoid memory leaks though, as that class stores all telemetry indefinitely. 

To stop the reports for now, I'd suggest changing this if statement: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/autorx/email_notification.py#L183

to:

```
                if (telemetry["alt"] < self.landing_altitude_threshold) and (telemetry["vel_v"] != -9999.0) and (
                    telemetry["vel_v"] < -2.0
                ):
```

Just remember you will need to stash these changes when I push out the fix.

ZigiWalter, 2021-04-06:
Cool, thanks again!

darksidelemm, 2021-04-18:
OK, there's a fix for this in the latest testing branch release (1.5.2-beta2), where the descent rate is now calculated within the email notification object from altitude/time deltas. It seems to work OK, and i've added in some code to clear out the data used after a few hours to avoid memory leaks (hopefully!). 

Let me know how it goes...

ZigiWalter, 2021-04-18:
Thanks!

Can you remind me - do we have somewhere instructions regarding how to work with the testing branch?
Or would "git checkout testing/master" be enough?

darksidelemm, 2021-04-18:
Yes, running 'git checkout testing' followed by 'git pull' should do it...

closed 2021-06-02 by darksidelemm