projecthorus/radiosonde_auto_rx #438: unsupported version of the Socket.IO or Engine.IO ??

K4KDR, 2021-04-06:
Hello!  Followed the update instructions for v1.5.1 and all "seemed" ok, but web display just shows "No scan data yet..."

I thought I was ok since the service runs normally:

```
pi@k4kdr-autoRX:~ $ sudo systemctl status auto_rx
● auto_rx.service - auto_rx
   Loaded: loaded (/etc/systemd/system/auto_rx.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-04-06 01:25:16 BST; 11min ago
 Main PID: 5949 (python3)
    Tasks: 13 (limit: 2181)
   CGroup: /system.slice/auto_rx.service
           └─5949 /usr/bin/python3 /home/pi/radiosonde_auto_rx/auto_rx/auto_rx.py -t 0
```

Checking /var/log/syslog, I see:

`The client is using an unsupported version of the Socket.IO or Engine.IO protocols`

... I have no idea what either of those is, but see something similar referenced in https://github.com/projecthorus/radiosonde_auto_rx/issues/424

My python3 is at:

`Python 3.7.3`

... is there some other piece of supporting software that I can add or update to get back up & running?

Many thanks!

-Scott,  K4KDR

K4KDR, 2021-04-06:
Update:

Installed npm with:

`sudo apt-get install npm`

... and using

`npm list socket.io`

... there did not appear to be any 'socket.io' installed.   Installed socket.io with:

`npm install socket.io`

... and no longer get that error.  Seeing no change in the web display, while not sure if it was necessary with this update, completed the **station.cfg.example** file and saved it with my settings as **station.cfg**

Service restarted and continues to look fine.  Even /var/log/syslog entries give the impression that all is well:

```
Apr  6 02:52:05 k4kdr-autoRX auto_rx[3852]: 2021-04-06 02:52:05,423 INFO:Scanner #0 - Running frequency scan.
Apr  6 02:52:36 k4kdr-autoRX auto_rx[3852]: 2021-04-06 02:52:36,532 INFO:Scanner #0 - Running frequency scan.
```

... however, the web interface still shows **Current Task: ???** and **No scan data yet...**

Any help appreciated.  Thanks!

darksidelemm, 2021-04-06:
A) did you update flask-socketio?
B) have you cleared your browser cache?

npm is unrealated to this application, we do not use npm, and instead have
the socketio js library in the repository.

73
Mark

K4KDR, 2021-04-06:
Thank you so much for the reply!  Even though I read the release notes and that is in BOLD, it did not register with me for some unknown reason.   Seems to be looking as it should now - thanks again!

closed 2021-04-06 by darksidelemm