projecthorus/radiosonde_auto_rx #439: RS41 decode of temp, humidity suspicious

mhaberler, 2021-04-06:
TLDR: the official data from the BUFR file has way more points with valid temperature and humidity (represented via dewpoint) than the auto_rx decode.

I've tracked serial R5140539 and have  logs from [sondehub](https://github.com/mhaberler/radiosonde-datacollector/blob/dev/bug/R5140539-fmt.json), [my own auto_rx install](https://github.com/mhaberler/radiosonde-datacollector/blob/dev/bug/20210406-023652_R5140539_RS41_403800_sonde.log), and the [official data distributed by DWD.de](https://raw.githubusercontent.com/mhaberler/radiosonde-datacollector/dev/bug/A_IUSD04LOWM060200_C_EDZW_20210406033800_39238684.json) (this is [my translation into GeoJSON which is more readable](https://github.com/mhaberler/radiosonde-datacollector/blob/dev/bug/A_IUSD04LOWM060200_C_EDZW_20210406033800_39238684.geojson))

let's take an example:

auto_rx decodes temp, hum as -273.0, -1.0:
```
timestamp,serial,frame,lat,lon,alt,vel_v,vel_h,heading,temp,humidity,pressure,type,freq_mhz,snr,f_error_hz,sats,batt_v,burst_timer,aux_data
2021-04-
06T02:37:08.000Z,R5140539,1180,46.98020,15.48561,3121.2,4.3,15.5,6.8,-273.0,-1.0,-1.0,RS41,403.800,16.7,0,8,2.9,-1,-1
```
now in my BUFR-derived geojson file we have for that sample perfectly legit temp and dewpoint values (pretty sure that was the same frame as the altitudes match exactly but the BUFR files have to frame numbers):
````
      {
            "type": "Feature",
            "geometry": {
                "type": "Point",
                "coordinates": [
                    15.48581,
                    46.98151,
                    3121.2
                ]
            },
            "properties": {
                "time": 1617676620,
                "gpheight": 3121,
                "temp": 261.24,
                "dewpoint": 259.9,
                "pressure": 676.3,
                "wind_u": 1.28,
                "wind_v": 14.64
            }
        },
````

@rs1729  @darksidelemm  not sure what to  do about it; if directed I could record IQ files if that helps.

flight & sonde details I could decode from the BUFR file:

Station: Graz-Thalerhof-Flughafen (WMO id: 11240)
location: 46.99396N 15.44706E
elevation: 338m
last recorded height: 20018m (53hPa)
text: Increasing pressure
samples: 193
track source: GPS
Synoptic time: 2021-04-06 02:00Z (9.7 hours old)
First seen: 2021-04-06 02:28Z
Last seen: 2021-04-06 03:34Z (67 min later)
Arrived: 2021-04-06 03:40Z (5 min after last seen)
Online: 2021-04-06 03:40Z (2 sec after arrival)

Sonde:
type: Vaisala RS41 with pressure derived from GPS height/ AUTOSONDE (Finland)
serial number: R5140539
transmit frequency: 403.8 MHz
SW version: MW41 2.15.0
humidity correction: Solar radiation and time lag correction provided by the manufacturer
pressure sensor: Derived from GPS
temperature sensor: Resistive sensor
humidity sensor: Humicap capacitance sensor with active de-icing method
humidity correction: Solar radiation and time lag correction provided by the manufacturer
geopotential height: calculated from GPS height
termination reason: Balloon Burst

Data reference:
source: GISC Offenbach
format: fm94/BUFR
source file: A_IUSD04LOWM060200_C_EDZW_20210406033800_39238684.bin
source archive: temp-fm94_20210406-034000653012_22314.zip
GTS topic: at/wien_met_com_centre/observation/upperair/land/fixed/temp/0-90n/90e-0
detail file format: version 7

TheSkorm, 2021-04-06:
I believe -273.0, -1.0 are placeholder values for when the decoder hasn't decoded enough frames to resolve the temperature and humidity values yet. You should start seeing real values after a handful of frames. The reason we had placeholders is due to the old habhub infrastructure requiring a value be entered regardless of if there was one.

mhaberler, 2021-04-06:
I see.. is it about incremental transmit of correction parameters?

darksidelemm, 2021-04-06:
Correct. The RS41 correction parameters are sent across many frames, and it can take quite a while to receive them, especially if you are not receiving every sequential frame.

closed 2021-04-06 by mhaberler