projecthorus/radiosonde_auto_rx #440: 1.5.1 web interface problems

gmke, 2021-04-08:
I have updated coming from 1.4 to 1.5.1 and followed the advice, including the flask-socketio update. I think the back end software runs without problems, including sending the emails, writing the log files etc. But the new web interface has problems and I did the shift-reload. Tried with Firefox and Chrome.
1) no display of Current Task (Current Task: ???)
2) no display of SDR in the SDR column
3) Age is always "old" although it is current data after a reload
4) no automatic reload/update of page, only after manual reload I see new data
5) the scan plot is always empty with message No scan data yet... although there is background scanning going on and new sondes are detected
Have I missed some important step?

snh, 2021-04-08:
Hey @gmke 👋 Thank you for the report. What browser and browser version are you running?

gmke, 2021-04-08:
Firefox and Chrome autoupdated to the latest version on Win10 20H2.

Markus

xscode, 2021-04-08:
Same issue as gmke here on windows 10 with chrome Version 89.0.4389.114 (Official Build) (64-bit).
When running directly in a terminal I get 
`2021-04-08 20:29:30,515 INFO:Telemetry Logger - Started Telemetry Logger Thread.
The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)
2021-04-08 20:29:31,127 ERROR:The client is using an unsupported version of the Socket.IO or Engine.IO protocols (further occurrences of this error will be logged with level INFO)
`
Not sure if that helps.

darksidelemm, 2021-04-08:
The errors seen above are a direct result of mismatched socketio versions on the client (browser) and server (auto_rx). I know Firefox really doesn't like to clear the cache for js files, so you may have to go and manually clear some of your browser cache for it to update.

It is possible that your flask-socketio version is still not up-to-date. Perhaps try running the command:
```
$ python3 -c "import flask_socketio; print(flask_socketio.__version__)"
```
In my case this returns '5.0.1'.

gmke, 2021-04-09:
I think I found the problem. Your test statement gave version 4.3.2. I checked again with "sudo pip3 install -U flask_socketio" and got the answer:

Requirement already up-to-date: flask_socketio in /usr/local/lib/python3.7/dist-packages (5.0.1)

After a second of confusion, I realized that there are two places for the python packages, depending on the user. I did a "pip3 install -U flask_socketio" and it installed version 5.0.1. The test statement now reports "python3 -c "import flask_socketio; print(flask_socketio.__version__)"
5.0.1"

Now the web interface works perfectly. The installation and update instructions should be updated to avoid mixing user and root installations. I'm not a python programmer and probably made a mistake a year ago.

darksidelemm, 2021-04-09:
Hmm, with the instructions run from scratch the flask-socketio package should be installed system-wide. 

Anyway, the docker image is probably the preferred away of avoiding these sorts of issues nowdays.

Glad to hear it's working!

closed 2021-04-09 by darksidelemm