projecthorus/radiosonde_auto_rx #452: Scan results query

mikethebee, 2021-04-22:
Should I be seeing "Scan Results" in the web section with 1.5.2-beta4? I did see all results previously. The terminal console reports seeing peaks while the web page shows _No scan data yet.._ It also shows _Current Task: ???_ which doesn't seem correct. It is detecting and uploading to habhub ok and shows the sonde on the map if I refresh. 

Anything I can check.

OS Lubuntu 18.04
FF 87.0
Python 3.6.9

LukePrior, 2021-04-22:
Can you try shift f5 on Chrome to force a complete page reload?

mikethebee, 2021-04-22:
Thanks for the suggestion, I had tried that. Also as the default IP changed from local nat to 0.0.0.0:5000 I had tried switching between them. The tracked sonde appeared, and the sdr log results, just seems to be the Current Task: and freq plot that are not updating.

steveKS, 2021-04-24:
https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.5.1

"You will need to update your flask-socketio version for the web interface to continue working. Upgrading can be performed by running sudo pip3 install -U flask-socketio"

This solved the problem for me.

mikethebee, 2021-04-26:
I updated pip and flask-socketio and that sorted it. Thanks for the suggestions.

closed 2021-04-26 by mikethebee