projecthorus/radiosonde_auto_rx #469: List previous sondes received by station on web interface

darksidelemm, 2021-04-30:
This one needs a bit of thought.

The idea would be to have some endpoint that allows parsing of the log folder, providing a list of previously seen radiosondes to the web interface. Then user can then select a sonde, which gets the back-end to load the log file and then perhaps the track is displayed on their map. Alternatively, a link to sondehub.org/card/ could be provided.

fkemps, 2021-04-30:
Hi there,
That would be more than welcome to review sondes from the past.
A nice integration would be to use (API integrate or recreate?) https://www.gpsvisualizer.com/. It shows you nicely the received beacons of the path.

My €2 cents,
Ferry

LukePrior, 2021-04-30:
Hey,

I was having a think about this and I think the first step would be to create this new function which can import past sonde from file. This function could then replace the website cache which would mean sonde would stay after update, the function can look at the existing archive_age[archive_age](https://github.com/projecthorus/radiosonde_auto_rx/blob/master/auto_rx/station.cfg.example#L396)  value and use this to limit the amount of sonde read in. We would need to ensure the function could load sonde from older files until the archive age is reached, we would also need to set a maximum amount of sonde to load into the interface without overloading the page.

This first step would set us up to loading historical sonde. The main question is how far back to look and how many sonde back to look for? 

I was thinking maybe the last 20 not currently in the interface or something close?

We could get the serial number and date/time of these sonde and add them all to a dropdown menu or similar possibly in the advanced control section. When the user selects one of the options it would be added to the table and drawn on the map.

I'm not sure how useful this will be as I would presume most people would prefer collated data from SondeHub for their historical data, if we get predictions added to old sonde in SondeHub I think linking there could prove sufficient. You could also go the route of importing data from SondeHub when loading a past sonde but I think that's out of the scope of this program.

darksidelemm, 2021-05-01:
I think we'll slate this one for 1.5.3, to avoid holding up 1.5.2 any longer.

darksidelemm, 2021-05-03:
I'm thinking maybe this should be a separate page to the main web interface.

Perhaps a page with a list of log files down the left, allowing the user to select one and request it be loaded in, with the data provided to the client for display.

LukePrior, 2021-05-03:
> I'm thinking maybe this should be a separate page to the main web interface.
> 
> Perhaps a page with a list of log files down the left, allowing the user to select one and request it be loaded in, with the data provided to the client for display.

If this is the route you want to go we should look into showing a telemetry graph like on SondeHub. We might just be able to copy the main page and remove parts not needed.

darksidelemm, 2021-05-03:
Yep, that would be a good idea. I probably wouldnt use the plotting system sondehub uses, maybe just use plotly to keep it simpler. I use that on sondehub.org/card/  https://github.com/projecthorus/sondehub-card/blob/main/index.html#L269

fkemps, 2021-05-03:
One use case would be to analyze first to last beacons received during the flight.
This allows you to see when you did receive sonde signals, for how long during the flight and at which height (the lower the better ;-) ).

I've seen a Google Earth KML plot showing this nicely in 3D.

LukePrior, 2021-05-04:
So I think the first step is replacing some functions to pull sonde data from the log files instead of local cache, to do this I think it would be best to create a few functions. We would first need one which could act as a replacement for `/get_telemetry_archive`, this could get the current system time and the `archive_age` from the configuration file and then subtract archive age to get how far back to go. The function could then iterate through the log files and check the file name to see if the date of the log file falls into the range. That function could then return the sonde data in the same format as we currently use to save changing the frontend code. This initial step would solve issues with the web history resetting after an update or power loss.

The next step would be creating a function that returns a list of all tracked sonde (we will probably need to put a limit on this not sure how many it could handle?), I think it would be good to display the launch time and sonde type, launch site would be good but I don't know any way of getting this information?

Finally, we would need a way to import the data for any sonde selected from the list.

In terms of how this would all be handled in the web interface, the first function would require no changes. I think having a separate page would be the best approach using the same main design as the homepage, we could add a new button to the main page settings which redirects the user. The left slide-out menu would display the list of historical sondes for the user to select while the right slide-out would remain a settings menu. I think the main page would display a map and telemetry graph, I like how all data is shown on one graph in SondeHub and when the user hovers over a certain point on the graph the map updates to show where the sonde was on its path at the time of that measurement. The settings menu would have options to show/hide the map and graph, dark/light mode selection, utc/local time, imperial/metric, these settings should be synced across both pages for a consistent experience.

In terms of other features I thought adding an option to download the log file for a sonde from the left slide-out menu would be useful and a way to view the data for that Sonde on SondeHub tracker or SondeHub database.

darksidelemm, 2021-05-04:
I'll have a go at adding some of the endpoints to do this on the weekend. 
Thoughts are:
- Endpoint which returns a list of the sondes in the log directory, ordered by time (newest to oldest).
- Endpoint which allows reading in of a log file (specified by Serial number I guess), and obtaining the sonde telemetry data as an object.

LukePrior, 2021-05-04:
> I'll have a go at adding some of the endpoints to do this on the weekend.

Sounds Good

> Endpoint which returns a list of the sondes in the log directory, ordered by time (newest to oldest).

It would be good if you could return serial, type, and time/date launched so that these can be displayed in list without making multiple requests. 

If you do this we could switch away from setting `archive_age` in the configuration file and instead make it a per-device option so that you can see more or less sonde on certain devices. The web interface could request the list and then determine which ones it wants to load and individually import them with the endpoint. This could allow us to set archive age in minutes or archive number in the number of sondes to return, just an idea.

> Endpoint which allows reading in of a log file (specified by Serial number I guess), and obtaining the sonde telemetry data as an object.

This sounds like the best option.

fkemps, 2021-05-04:
Launch sites and detailed information can be found online at https://radiosondy.info/index.php?.

See the dropdown list "Launch Site".
Another page listing them including details e.g. LAT/LON, frequencies on https://radiosondy.info/sonde_table.php?table=meteo_service&ListDown=1

darksidelemm, 2021-05-04:
I don't really intend to include that kind of information in auto_rx itself.

fkemps, 2021-05-04:
Sure, it was more to answer LukePrior's question "launch site would be good but I don't know any way of getting this information?"

fkemps, 2021-05-16:
I want to give my compliments, the new historical data on the testing branch looks great!

Most weather balloon hunters are interested in the lowest altetude recordings. Could you consider adding that column right of count?

Optional: longest distance, but think that data isn't available by default and needs calculation.

darksidelemm, 2021-05-16:
This data is actually now fairly easily available, but it will make the table a lot wider!

LukePrior, 2021-05-16:
Yeah currently the table uses fixed values if we were to add more columns we would have to adjust the styling and/or add the option to show/hide columns like on the main page. I'm not interested in adding this functionality but the code is available in index.html.

closed 2021-06-02 by darksidelemm

fkemps, 2021-06-17:
Hi Luke,

I noticed that the "Plot Landings" is not presenting the actual landing place but the last received signal location.
Would "Plot last beacon", "Plot Last destination" or alike be more correct?

A column with "latest height" on the table is still very welcome.
If table space is an issue, may I suggest limiting wide on Type for "Intermed Systems iMet-1/4" and make it "IS iMet-1/4"?
We could as well trim the leading 4 spaces on all the columns to 1 ;-)

For sonde hunters, the last received signal with the lowest height is an important factor while following/tuning the receiver setup and auto_rx. 

What do you think?
Regards,
Ferry

darksidelemm, 2021-06-18:
Changing Plot Launches/Landings to Plot Firsts / Lasts is easy enough and is definitely more correct.

The other one is doable, we now have the data available, and it looks like there is probably enough space that could be recovered from making some of the other columns a little narrower. Unsure how it will go on non-desktop platforms.