projecthorus/radiosonde_auto_rx #483: Why doesn't my Pi 4 image work on my PI 3B+ ?

m1ctk, 2021-05-09:
So I remove the SD card from my Pi 4 into my Pi 3B+ and I get no reception at all, like the antenna is not connected. Any ideas please what might be wrong ? rtl_test shows the V3 dongle no problem

darksidelemm, 2021-05-09:
This is a Raspbian issue, not an auto_rx issue.

I know there are some architecture differences between the RPi 3 and RPi 4, so i'm not surprised a SD card wouldn't completely work when switched between them.

I expect you are going to need to re-build the SD card.

closed 2021-05-09 by darksidelemm

m1ctk, 2021-05-09:
Sorry it was late I will delete this thread but yes some architectural differences must be the reason for the fail. I will do get another image for the pi 3. Thanks