projecthorus/radiosonde_auto_rx #512: Problem with rx-1.5.3

xtronics, 2021-05-25:
May 25 13:26:00 sonde systemd[1]: Stopped auto_rx.
May 25 13:26:00 sonde systemd[1]: Starting auto_rx...
May 25 13:26:05 sonde systemd[1]: Started auto_rx.
May 25 13:26:06 sonde auto_rx[17758]: Traceback (most recent call last):
May 25 13:26:06 sonde auto_rx[17758]:   File "/usr/src/radiosonde_auto_rx/auto_rx/auto_rx.py", line 26, in <module>
May 25 13:26:06 sonde auto_rx[17758]:     from autorx.scan import SondeScanner
May 25 13:26:06 sonde auto_rx[17758]:   File "/usr/src/radiosonde_auto_rx/auto_rx/autorx/scan.py", line 18, in <module>
May 25 13:26:06 sonde auto_rx[17758]:     from .utils import (
May 25 13:26:06 sonde auto_rx[17758]:   File "/usr/src/radiosonde_auto_rx/auto_rx/autorx/utils.py", line 20, in <module>
May 25 13:26:06 sonde auto_rx[17758]:     import semver
May 25 13:26:06 sonde auto_rx[17758]: ModuleNotFoundError: No module named 'semver'
May 25 13:26:06 sonde systemd[1]: auto_rx.service: Main process exited, code=exited, status=1/FAILURE
May 25 13:26:06 sonde systemd[1]: auto_rx.service: Failed with result 'exit-code'.

F1TKE, 2021-05-25:
**Note: If you are using a 'native' install of auto_rx, you will need to install the 'semver' Python package by running: `sudo pip3 install semver`**

 

Cet email a fait l'objet d'une analyse antivirus par AVG.
http://www.avg.com

darksidelemm, 2021-05-25:
Please make sure you read the release notes when updating. This particular issue was mentioned in bold text halfway down: https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.5.3

closed 2021-05-25 by darksidelemm

xtronics, 2021-05-26:
One more reason to simply make a deb out of this ..  I had seen the version was out-of-date in the logs - so I just updated and rebuilt (the build scrip might want to notice the missing dependency).  If you use the very mature Deb infrastructure - it even has a way to display release notes during updates. 

Making it into a Deb is a bit of a hassle - but once in the system, there are even package-lint tools that checks the package for problems.

TheSkorm, 2021-05-26:
Issue : #2

Apart from .deb's being a pain to build in the first place, there are some things that would need to be fixed in autorx to make a deb more worthwhile:
 - move logs to /var/log
 - move config to /etc
 - make where autorx is installed to not need to be written
 - package encoders prebuilt

but it does need someone to do this work. The other question would be is if .deb is the right thing to focus on, a lot of users run it on non debian machines (like myself) where a pypi package would make more sense.

darksidelemm, 2021-05-28:
I would prefer not to package auto_rx as a .deb (or .rpm, or whatever). 
I agree that pypi would make more sense, but we have a lot of different binaries to build to make it work, and I'm unsure how that works with pypi.

IMO the docker container solves a lot of these issues, and also helps resolve dependency issues - going to something like a deb isn't always going to solve this well - if the dependency (or more importantly, the particular version of the dependency we require) isn't available from that distro's package manager, then what?