projecthorus/radiosonde_auto_rx #513: Sondes M20

F21467, 2021-06-01:
around Toulouse, we can see sondes M20 but no data are read ?

darksidelemm, 2021-06-01:
Is auto_rx detecting the sonde at all? Are there any errors shown in the log?

darksidelemm, 2021-06-01:
Ahh I think you mean *sensor* data. I've just turned on a M20 sonde here and confirmed it is decoded and uploads to sondehub fine, however there is no support for decoding sensor data for M20 sondes.
I'm not sure what the state of temp/humidity decoding on the M20 is, I don't think any other decoders support it yet? Any ideas @rs1729 ?

rs1729, 2021-06-02:
I believe the sensors are similar to M10. I was looking into the raw sensor data some time ago, but didn't have the time to compare it to published sounding data then. Probably I will have a little more time the next weeks and look at the temperature again.

F21467, 2021-06-02:
I am not competent enough to be able to make a relevant contribution, I now know that this is not a fault with my installation, thank you for your help, I will be patient

closed 2021-06-02 by F21467