projecthorus/radiosonde_auto_rx #518: WebSocket transport is not available

F80543, 2021-06-03:
Good morning from France,
Whether I start from a fresh Raspberry Pi OS MicroSD card or I go through an update (card with V1.5.1 or V1.5.2), installation of version 1.5.3 goes correctly and my data are transmitted to Sondehub and APRS-IS.

However, when I open localhost: 5000, I get the message:
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO).

A Google search tells me that the alert would come from :
https://github.com/miguelgrinberg/python-engineio/blob/main/engineio/server.py (ligne 588)

How can I solve this problem (if there is a problem) ?

Thanks,
Francois (F80543)

My configuration :

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
$ python3 --version
Python 3.7.3

darksidelemm, 2021-06-03:
Can you check that the web interface still works? 

This message indicates that the webserver we are using can't start a websocket session, so it'll fall back to using 'short polling' when communicating with clients that connect to the auto_rx web interface.

This isn't ideal, but for just a few clients, it's still workable, and the web interface should still work as normal.

F80543, 2021-06-04:
I am using Chromium Version 88.0.4324.187 (Build officiel).
I tested the websocket using https://stackoverflow.com/questions/62755998/websocket-does-not-connect-to-a-specific-host-until-chromium-is-restarted and all is OK.
The web interface works fine (setting tab at right, log tab at left…).

This morning, I installed Firefox 78.11.0esr (32-bit). I receive the same message.
**Like with Chromium, this message only appears (in duplicate) once: when you open the internet browser for the first time.**

Hereafter the messages where I replaced my APRS data with [--XXX--]
--------------
Lancement de Radiosonde_auto_rx
2021-06-04 09:36:04,981 INFO:Reading configuration file...
2021-06-04 09:36:07,919 INFO:Config - Tested SDR #0 OK
2021-06-04 09:36:07,926 INFO:Started Flask server on http://0.0.0.0:5000
2021-06-04 09:36:07,928 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2021-06-04 09:36:09,271 INFO:APRS-IS - Connected to APRS-IS server radiosondy.info:14590
2021-06-04 09:36:09,273 INFO:APRS-IS - Uploaded to APRS-IS: [--XXX--]>APRARX,SONDEGATE,TCPIP,qAR,[--XXX--]`radiosonde_auto_rx SondeGate v1.5.3 !wj!!
2021-06-04 09:36:09,273 INFO:APRS-IS - APRS Uploader Started.
2021-06-04 09:36:09,274 INFO:OziMux - Started OziMux / Payload Summary Exporter
2021-06-04 09:36:09,275 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2021-06-04 09:36:09,614 INFO:Version - Local Version: 1.5.3 - Up to date!
2021-06-04 09:36:09,615 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2021-06-04 09:36:09,951 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
2021-06-04 09:36:12,570 INFO:Scanner #0 - Starting Scanner Thread
2021-06-04 09:36:12,577 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:36:43,214 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:37:13,221 INFO:Scanner #0 - Running frequency scan.
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-06-04 09:37:21,468 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-06-04 09:37:24,217 INFO:Flask - New Web Client connected!
2021-06-04 09:37:43,167 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:38:13,280 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:38:43,164 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:39:13,210 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:39:43,290 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:40:13,300 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:40:43,130 INFO:Scanner #0 - Running frequency scan.
2021-06-04 09:40:50,618 INFO:Flask - New Web Client connected!
2021-06-04 09:41:13,357 INFO:Scanner #0 - Running frequency scan.

mecrocket, 2021-06-15:
Same issue here after instaling on a Ubuntu desktop 20.04.2 ...
Error appear in chasemapper (1.3.1)  and auto_rx (1.5.3)...
I test it also using the docker images and the error was the same ...
I test it from Ubuntu Chromium and Windown 10 Chrome, both up to date in version 91.0.4472.101 (Official Build) (64-bit)

darksidelemm, 2021-06-15:
Is the web interface still functional?

F80543, 2021-06-15:
darksidelemm asked : 
> Is the web interface still functional?

For me (Raspberry Pi OS, last version) : Yes for radiosonde_auto_rx and Yes for chasemapper.

mecrocket, 2021-06-15:
> Is the web interface still functional?

In Chasemapper, the panels and controls works normally but, the log panel (Log Messages) in the web interface are not showing anything at the same time that the log file (chasemapper/log_files/) show all incoming positions (via UDP) correctly... 
Usually (not always), after force reloading the page, you get a new position and the balloon is positioned correctly but never updated again ...

In auto_rx, the web interface show rigth info one time and never again until you for Crhome to reload page ... Interface

I did a fresh install of both projects following instructions steps one by one  in a new Ubuntu 20.04 just installed in a laptop.
Thanks in advance

ZigiWalter, 2021-07-10:
Facing a a similar issue here, on a secondary RPI4 I'm running occasionally. I have recently "sudo apt full-upgrade" it, but not sure it's related. When connecting with a browser, the log says:

`2021-07-10 20:44:04,850 INFO:Scanner #0 - Waiting for current scan to finish...
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-10 20:44:05,906 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-10 20:44:06,294 INFO:Flask - New Web Client connected!
`

The GuI on port 5000 is loading and seems to be working  just fine - not sure what is broken, if at all. Already tried - clearing the browser cache, switching to a different browser.

BTW, I see a similar error message in the logs at https://github.com/projecthorus/radiosonde_auto_rx/issues/538.

In case needed, I've pasted some version info below.

Thanks.

```
pi@RPI4B:~ $ cat /etc/debian_version
10.10

pi@RPI4B:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

pi@RPI4B:~ $ uname -a
Linux RPI4B 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux

pi@RPI4B:~ $ python3 -c "import websocket; print(websocket.__version__)"
0.57.0
```

darksidelemm, 2021-07-10:
Yep it seems it's just a warning, and the web interface continues to work normally. This is occurring because the webserver in use by auto_rx (and socketio, and flask) doesn't actually support websockets correctly, and so the socketio messaging transport drops down to a 'short polling' method of communication. Not good for lots of users, but fine for the small number that are normally connected to a auto_rx web interface.

I'm a little bit hesitant to suppress the warning (if I even can), as it may also suppress any other warnings that we may actually care about.

F80543, 2021-08-30:
There is no error/warning message with the v.1.5.5.
Francois,

Torrehoradada, 2023-05-21:
I did a fresh install of Radiosonde Auto-RX 1.6.1 onto my Linux Mint 21.1
Everything seems to work quite good, altheough the local web server seems to work a little slow.
Each time fireing auto_rx.sh and each time I shut down I get the same nasty ERROR message. There is no hint in any documentation what I can do against it, so it seems this is a quirk in radiosonde auto_rx itself.
It seems the web server does not work like expected.
For reference this is the log output:

> /radiosonde_auto_rx/auto_rx$ ./auto_rx.sh
> 2023-05-21 21:55:02,874 INFO:Reading configuration file...
> 2023-05-21 21:55:04,948 INFO:Config - Tested RTLSDR #0 OK
> 2023-05-21 21:55:04,949 INFO:Started Flask server on http://0.0.0.0:5000
> 2023-05-21 21:55:04,949 INFO:Telemetry Logger - Started Telemetry Logger Thread.
> The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
> 2023-05-21 21:55:05,593 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
> 2023-05-21 21:55:05,626 INFO:Flask - New Web Client connected!

Would be nice to get this error corrected...

darksidelemm, 2023-05-21:
The error message is there because we don't install any libraries alongside flask which supports true Websocket mode. 
The web interface should fall back to fast-polling mode, which will still make the page work.

The things to look for to confirm the web interface is doing something useful is:
- Log messages appearing in the log area as scans occur.
- The Scan plot being updated occasionally
- When receiving a radiosonde, telemetry and positions updating.

If you don't see *any* of the above, then you can look at your browsers javascript console (often somewhere in Developer Tools) to see if there are any other errors on page loadl

Torrehoradada, 2023-05-21:
Hi Mark, thanks for your quick response!
Well, -I'm no expert in these things, but I cannot fully agree. As much as I understand, Flask is the internal web server in auto_rx.
If this drops an error message, -well, I understand there is an error. That's an instinct!
Maybe some library missing, but this library must be called to be missed... maybe it HAS TO be installed...?
Or probably a configuration error of Flask? Or a buggy web server at all? Wouldn't it be wise to use one which runs without errors?
Sorry for being such intense, but my feeling is that error messages do always have a meaning, otherwise they wouldn't be implemented.
-And yes I fear error messages...
Have a nice day!
Gunther.

darksidelemm, 2023-05-21:
This error message been known about for a while. It is mentioned in the release notes: https://github.com/projecthorus/radiosonde_auto_rx/releases/tag/v1.6.1

There may be new libraries which will provide the websocket functionality. 

Either way, can you do the checks I mentioned above to determine if the web interface is working.

Torrehoradada, 2023-05-21:
As I wrote before, it seems to work. The data is displaying, sometimes with quite a delay, but they come. Especially trace lines come up truncated and grow pixel by pixel sometimes... After reload everything is there but can get delays again later.
Btw: I use Firefox "mint-001 - 1.0" which is a version 113.0.1 (64-Bit).

darksidelemm, 2023-05-21:
That kind of issue doesn't sound like a Websockets issue. It sounds like the data is making it to the browser, but it's not being displayed correctly. A screenshot of that 'truncation' issue would be interesting.

Torrehoradada, 2023-05-22:
Hi, it took a while to repeat, -needed to see a sonde flying.
Ich appeares when zooming in and/or out, or shifting the map display. I got two examples made short after the other as you can see from the time stamps. In the third picture you see the complete part of the track.

[image]
[image]
[image]

Actually the track is much longer, but the effect is only seen on shoft pieces. Also the track line grows in the reverse direction very slowly, in the end it will reach the map boundaries.

darksidelemm, 2023-05-22:
OK, well unfortunately this is likely an issue with the Leaflet mapping library we are using, other than that I don't know what the cause would be...

darksidelemm, 2023-05-25:
To resolve the websocket error, you can try installing the simple-websocket python library.

If running a native install:
```
$ sudo pip3 install simple-websocket
```
the restart auto_rx.

If running via docker, switch to the testing branch and update that.

closed 2023-06-30 by darksidelemm