projecthorus/radiosonde_auto_rx #525: Failure to recover from almanac download error

dbrooke, 2021-06-21:
I noticed that no sondes had been received for a while. On investigation the web UI was showing as if it was still decoding a sonde from several days earlier.

Attached is the relevant section of log where it can be seen that an almanac download fails after which auto_rx appears to try to shut down.

With regard to "Failed to parse: :::5000", the web UI is configured to listen on IPv6 address with "web_host = ::" and this usually works as can be seen at the end of the log where I restarted auto_rx.

[auto_rx.log](https://github.com/projecthorus/radiosonde_auto_rx/files/6685983/auto_rx.log)

darksidelemm, 2021-06-21:
OK, so the error regarding GPS data grabbing is here:
```
Jun 17 08:59:37 fowler auto_rx[322]:   File "/home/pi/radiosonde_auto_rx/auto_rx/autorx/gps.py", line 81, in get_almanac
Jun 17 08:59:37 fowler auto_rx[322]:     logging.error("GPS Grabber - Failed to download almanac data - " % str(e))
Jun 17 08:59:37 fowler auto_rx[322]: TypeError: not all arguments converted during string formatting
```

Definitely an error in the code, which I'll fix in the testing branch shortly

darksidelemm, 2021-06-21:
This should now be fixed in the testing branch.

darksidelemm, 2021-07-04:
Fixed in 1.5.4

closed 2021-07-04 by darksidelemm