projecthorus/radiosonde_auto_rx #535: Feature Request:  Place callsign in window title for web client.

Moll1989, 2021-07-04:
A few of us log in to multiple stations, sometimes simultaneously.

It would be good if the page title showed the station callsign so that we can see which tab is which in our browser, rather than all titles being "Radiosonde Auto-RX Status".  For Example something like "VK5CBM_AUTO_RX Status" would make it easy to differentiate tabs.

It looks like this could fairly simply be included in the index.html template?

darksidelemm, 2021-07-05:
This should be fairly trivial to do...

darksidelemm, 2021-07-05:
Added this in the testing branch.

F1TKE, 2021-07-05:
Nice 
[image]

darksidelemm, 2021-07-05:
Currently it's using both the sondehub and APRS callsigns ( SONDEHUB / APRS ) but I might wind that back to just use the Sondehub callsign...

closed 2021-08-15 by darksidelemm