projecthorus/radiosonde_auto_rx #536: Strange polygon on coverage map

g7gpr, 2021-07-05:
[image]

darksidelemm, 2021-07-05:
That is a bit odd, and is probably an artefact of not having enough datapoints.

If you could zip up your log directory and post that somewhere, that would help figure out what's going on.

g7gpr, 2021-07-05:
[log.tar.gz](https://github.com/projecthorus/radiosonde_auto_rx/files/6762913/log.tar.gz)

Log directory attached. Great new features.

LukePrior, 2021-07-05:
> That is a bit odd, and is probably an artefact of not having enough datapoints.

Yeah it's exactly that not much can really be done with the existing code to fix it, would need some complex algorithm or something

darksidelemm, 2021-07-05:
Actually something that might be a simple change is to insert the station position when there is no data for a particular azimuth. That would probably clean up the example above.

g7gpr, 2021-07-06:
I'm sure you are already across this, but S3551090 is the sonde that causes the issue. This was a rare reception of a sonde from Geraldton, rather than Perth. Once I rename that file the polygon is normal.

darksidelemm, 2021-08-15:
This should be fixed in 1.5.5

g7gpr, 2021-08-15:
[image]

I believe it is. I will test again tomorrow with the log directory above.

g7gpr, 2021-09-02:
[image]

I untarred log.tar.gz file from the example, and reran the coverage - looks perfect now.

closed 2021-09-02 by g7gpr