projecthorus/radiosonde_auto_rx #538: No data from RTLSDR V3

cybersyx, 2021-07-06:
Hi everyone, I have a problem with a fresh install of auto_rx. RTLSDR V3 SDR key on raspberry. Auto_rx installed as per wiki instructions but when I start the python3 auto_rx.py I only get this log where it indicates that it finds a peak on the frequencies and nothing more.

pi @ raspberrypi-sdr2: ~ / radiosonde_auto_rx / auto_rx $ python3 auto_rx.py
2021-07-06 14: 57: 53,586 INFO: Reading configuration file ...
2021-07-06 14: 57: 53,619 WARNING: Config - Did not find save_raw_hex setting, using default (disabled)
2021-07-06 14: 57: 56,746 INFO: Config - Tested SDR # 0 OK
2021-07-06 14: 57: 56,753 INFO: Started Flask server on http://0.0.0.0:5000
2021-07-06 14: 57: 56,755 INFO: Telemetry Logger - Started Telemetry Logger Thread.
2021-07-06 14: 57: 57.842 INFO: APRS-IS - Connected to APRS-IS server radiosondy.info:14580
2021-07-06 14: 57: 57.845 INFO: APRS-IS - Uploaded to APRS-IS: IU0DZJ-14> APRARX, SONDEGATE, TCPIP, qAR, IU0DZJ-14:! 4124.28N / 01320.94E`IU0DZJ ALTITUDE radiosonde_auto_rx SondeGate .5.4! W3 !!
2021-07-06 14: 57: 57.847 INFO: APRS-IS - APRS Uploader Started.
2021-07-06 14: 57: 57,749 INFO: OziMux - Started OziMux / Payload Summary Exporter
2021-07-06 14: 57: 57.853 INFO: Sondehub Uploader - Started Sondehub Uploader Thread.
2021-07-06 14: 57: 58,361 INFO: Version - Local Version: 1.5.4 - Up to date!
2021-07-06 14: 57: 58,363 INFO: Task Manager - SDR # 0 has been allocated to Scanner.
2021-07-06 14: 57: 58,578 INFO: Sondehub Uploader - Uploaded station information to Sondehub.
2021-07-06 14: 58: 01,468 INFO: Scanner # 0 - Starting Scanner Thread
2021-07-06 14: 58: 01,477 INFO: Scanner # 0 - Running frequency scan.
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-06 14: 58: 01,666 ERROR: The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-06 14: 58: 01,736 INFO: Flask - New Web Client connected!
2021-07-06 14: 58: 22,488 INFO: Scanner # 0 - Detected peaks on 5 frequencies (MHz): [402.7 404.2 403.2 402.28 402.]
2021-07-06 14: 59: 25,831 INFO: Scanner # 0 - Running frequency scan.
2021-07-06 14: 59: 46,482 INFO: Scanner # 0 - Detected peaks on 5 frequencies (MHz): [402.7 404.2 403.2 402.28 402.]

I also tried to use the sdr key with rtl_tcp on sdrsharp and it works correctly.

I want to clarify that the same configuration works perfectly on another raspberry with the same sdr key. Thank you all

darksidelemm, 2021-07-06:
The fact that you are getting peaks (in particular the 403.2 MHz birdie common to the RTLSDR v3) suggests that auto_rx is talking to the RTLSDR just fine. If you enable the scan plot from the web interface you'll see the results of this.

cybersyx, 2021-07-06:
yes there were radiosondes in flight. Even now they are there but it does not detect them.

darksidelemm, 2021-07-07:
Can you post a picture of the scan plot output? Is there any other log
entries?

Maybe try running autorx up manually with the -v option to get debug output
to see what's being called.

cybersyx, 2021-07-07:
[image]
This is the scan plot and this is the log with -v option for debug:

2021-07-07 13:45:50,724 INFO:Reading configuration file...
2021-07-07 13:45:50,756 WARNING:Config - Did not find save_raw_hex setting, using default (disabled)
2021-07-07 13:45:50,937 DEBUG:lsusb parsing error: unexpected indentation: "           ** UNAVAILABLE **"
2021-07-07 13:45:53,871 INFO:Config - Tested SDR #0 OK
2021-07-07 13:45:53,878 INFO:Started Flask server on http://0.0.0.0:5000
2021-07-07 13:45:53,883 INFO:Telemetry Logger - Started Telemetry Logger Thread.
2021-07-07 13:45:54,055 DEBUG:APRS-IS - Logging in: user IU0DZJ-14 pass 20665 vers VK5QI-AutoRX

2021-07-07 13:45:54,056 DEBUG:APRS-IS - Setting Filter: #filter p/ZZ

2021-07-07 13:45:54,057 DEBUG:APRS-IS - Setting Filter: #filter -t/po

2021-07-07 13:45:55,059 DEBUG:APRS-IS - Server Logon Response: # aprsc 2.1.8-gf8824e8
# logresp IU0DZJ-14 verified, server SQ6KXY-1
# filter p/ZZ active
# filter -t/po active
2021-07-07 13:45:55,060 INFO:APRS-IS - Connected to APRS-IS server radiosondy.info:14580
2021-07-07 13:45:55,062 DEBUG:APRS-IS - Started APRS Uploader Thread.
2021-07-07 13:45:55,065 INFO:APRS-IS - Uploaded to APRS-IS: IU0DZJ-14>APRARX,SONDEGATE,TCPIP,qAR,IU0DZJ-14:!4124.28N/01320.94E`IU0DZJ ALTITUDE radiosonde_auto_rx SondeGate v1.5.4 !w3!!
2021-07-07 13:45:55,066 INFO:APRS-IS - APRS Uploader Started.
2021-07-07 13:45:55,068 INFO:OziMux - Started OziMux / Payload Summary Exporter
2021-07-07 13:45:55,072 INFO:Sondehub Uploader - Started Sondehub Uploader Thread.
2021-07-07 13:45:55,587 INFO:Version - Local Version: 1.5.4 - Up to date!
2021-07-07 13:45:55,588 INFO:Task Manager - SDR #0 has been allocated to Scanner.
2021-07-07 13:45:55,746 DEBUG:lsusb parsing error: unexpected indentation: "           ** UNAVAILABLE **"
2021-07-07 13:45:55,880 INFO:Sondehub Uploader - Uploaded station information to Sondehub.
The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-07 13:45:58,267 ERROR:The WebSocket transport is not available, you must install a WebSocket server that is compatible with your async mode to enable it. See the documentation for details. (further occurrences of this error will be logged with level INFO)
2021-07-07 13:45:58,696 INFO:Scanner #0 - Starting Scanner Thread
2021-07-07 13:45:58,710 INFO:Scanner #0 - Running frequency scan.
2021-07-07 13:45:58,712 DEBUG:Scanner #0 - Running command: timeout -k 30 30 rtl_power -f 400000000:406000000:800 -i 20 -1 -c 20% -p 0 -d 0 log_power_0.csv
2021-07-07 13:46:00,444 INFO:Flask - New Web Client connected!
2021-07-07 13:46:19,358 INFO:Scanner #0 - Detected peaks on 5 frequencies (MHz): [402.7  404.2  403.2  402.28 402.  ]
2021-07-07 13:46:19,361 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 402700000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:46:19,362 DEBUG:Scanner #0 - Attempting sonde detection on 402.700 MHz
2021-07-07 13:46:29,914 DEBUG:Scanner - dft_detect exited in 10.5 seconds with return code 1.
2021-07-07 13:46:29,921 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 404200000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:46:29,923 DEBUG:Scanner #0 - Attempting sonde detection on 404.200 MHz
2021-07-07 13:46:31,073 DEBUG:APRS-IS - Incoming data from APRS-IS: # aprsc 2.1.8-gf8824e8 7 Jul 2021 12:46:13 GMT SQ6KXY-1 188.121.31.198:14580

2021-07-07 13:46:40,672 DEBUG:Scanner - dft_detect exited in 10.7 seconds with return code 1.
2021-07-07 13:46:40,675 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 403200000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:46:40,678 DEBUG:Scanner #0 - Attempting sonde detection on 403.200 MHz
2021-07-07 13:46:51,325 DEBUG:Scanner - dft_detect exited in 10.6 seconds with return code 1.
2021-07-07 13:46:51,328 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 402280000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:46:51,331 DEBUG:Scanner #0 - Attempting sonde detection on 402.280 MHz
2021-07-07 13:47:01,045 DEBUG:APRS-IS - Incoming data from APRS-IS: # aprsc 2.1.8-gf8824e8 7 Jul 2021 12:46:33 GMT SQ6KXY-1 188.121.31.198:14580
# aprsc 2.1.8-gf8824e8 7 Jul 2021 12:46:53 GMT SQ6KXY-1 188.121.31.198:14580

2021-07-07 13:47:02,008 DEBUG:Scanner - dft_detect exited in 10.7 seconds with return code 1.
2021-07-07 13:47:02,011 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 402000000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:47:02,014 DEBUG:Scanner #0 - Attempting sonde detection on 402.000 MHz
2021-07-07 13:47:07,504 INFO:Flask - New Web Client connected!
2021-07-07 13:47:12,836 DEBUG:Scanner - dft_detect exited in 10.8 seconds with return code 1.
2021-07-07 13:47:12,839 DEBUG:Scanner #0 - No sondes detected.
2021-07-07 13:47:22,852 INFO:Scanner #0 - Running frequency scan.
2021-07-07 13:47:22,854 DEBUG:Scanner #0 - Running command: timeout -k 30 30 rtl_power -f 400000000:406000000:800 -i 20 -1 -c 20% -p 0 -d 0 log_power_0.csv
[... cut ...]
2021-07-07 13:49:28,784 DEBUG:Scanner - dft_detect exited in 10.7 seconds with return code 1.
2021-07-07 13:49:28,786 DEBUG:Scanner #0 - Using detection command: timeout 10s rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 403200000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 13:49:28,789 DEBUG:Scanner #0 - Attempting sonde detection on 403.200 MHz
2021-07-07 13:49:31,005 DEBUG:APRS-IS - Incoming data from APRS-IS: # aprsc 2.1.8-gf8824e8 7 Jul 2021 12:49:13 GMT SQ6KXY-1 188.121.31.198:14580

darksidelemm, 2021-07-07:
The detection stags appear to be running OK, though they do seem to be taking a bit longer to run that I expect (hitting the overall timeout for the command). 

What platform is this running on?

darksidelemm, 2021-07-07:
As an example from my station:
```
2021-07-07 21:56:10,303 INFO:Scanner #00000002 - Detected peaks on 2 frequencies (MHz): [401.5 402.5]
2021-07-07 21:56:10,305 DEBUG:Scanner #00000002 - Using detection command: timeout 10s /home/pi/rtl-sdr/build/src/rtl_fm -p 0 -d 00000002 -g 32.8 -M raw -F9 -s 48000 -f 401500000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 21:56:10,306 DEBUG:Scanner #00000002 - Attempting sonde detection on 401.500 MHz
2021-07-07 21:56:18,223 DEBUG:Scanner - dft_detect exited in 7.9 seconds with return code 1.
2021-07-07 21:56:18,226 DEBUG:Scanner #00000002 - Using detection command: timeout 10s /home/pi/rtl-sdr/build/src/rtl_fm -p 0 -d 00000002 -g 32.8 -M raw -F9 -s 48000 -f 402500000 2>/dev/null |./dft_detect -t 5 --iq --bw 20 --dc - 48000 16 2>/dev/null
2021-07-07 21:56:18,227 DEBUG:Scanner #00000002 - Attempting sonde detection on 402.500 MHz
2021-07-07 21:56:26,190 DEBUG:Scanner - dft_detect exited in 8.0 seconds with return code 1.
2021-07-07 21:56:26,192 DEBUG:Scanner #00000002 - No sondes detected.
```
That the detection stage is taking so long makes me wonder if you'r hitting a CPU limit, and so samples are being dropped.

cybersyx, 2021-07-07:
[image]

This is the scan plot.

cybersyx, 2021-07-07:
> The detection stags appear to be running OK, though they do seem to be taking a bit longer to run that I expect (hitting the overall timeout for the command).
> 
> What platform is this running on?

The platform is Raspberry Pi 3 Model B Plus.

darksidelemm, 2021-07-07:
That should be working just fine. To check you are not being affected by the rtl_fm CPU usage bug that seems to be an issue at the moment, try running (in a terminal):
```
rtl_fm -p 0 -d 0 -M raw -F9 -s 48000 -f 404200000 > /dev/null
```
.. and then in another terminal run 'top'.

If rtl_rm is using more than ~15% CPU then you're being affected by the current rtl_fm bug and probably need to go install rtl-sdr from source: https://github.com/projecthorus/radiosonde_auto_rx/wiki#421-installing-rtl-sdr-from-source-if-required
(Or switch to the Docker image, which solves this problem too)

darksidelemm, 2021-07-07:
Something else - I note that you have a few always_scan frequencies set (402.7 404.2 at a guess), and there are no corresponding peaks at those frequencies in your scan plot. It's quite possible the sondes you are looking for are just not in the air, or you are not receiving them sufficiently strongly for decoding to be possible. The peaks in your scan plot may also just be birdies.

cybersyx, 2021-07-08:
I confirm that there are radiosondes on that frequency in fact with another receiver I receive and decode them without problems. I tried to replicate the exact same configuration on another rapberry with another sdr rtl key and a different antenna but the result is the same. I can't really understand what the problem could be. I’m alreadt using the rtl_sdr versione from source. Thanks for the support .

darksidelemm, 2021-07-08:
Given that there are no obvious errors, I'm not sure how to proceed with
this one, sorry.

darksidelemm, 2021-07-08:
You mention 'another receiver'. Perhaps try with the RTLSDR and antenna you
are trying to use with auto_rx. Perhaps there are performance differences.

cybersyx, 2021-07-08:
I'll explain. I have three receivers in total. One at my qth (raspberry pi4 with rtl_sdr v3, no problem) and two in a location at 1500 meters (two raspberry pi3 with two rtl_sdr v3 and two 400 mhz monoband antennas). The receiver at my qth (configured about a month ago) works perfectly, while the two rpi3 configured in the same way not even one works. At this point I don't think it could be a hardware problem as they both don't decode anything.

darksidelemm, 2021-07-08:
Sure, but from your logs it looks like everything is running just fine. The only real possibilities now are:
- CPU is being overloaded resulting in dropped samples and no detections. Checking `top` while running the rtl_fm command I mentioned earlier can test this.
- Poor signal, such that sondes cannot be detected. Comparing scan plots when a sonde is in the air (noting that the scan plot will not update once decoding has started) may show what's going on with this.  In particular, compare what you have shown above with what you see when a sonde is in the air. Perhaps restart your working station and compare the scan plots on that.

Unless you can 100% confirm that that particular RTLSDR + that particular antenna is receiving a signal using some other application (gqrx via rtl_tcp as an example, if there's enough bandwidth), then all I can think of right now is a hardware issue.

closed 2021-12-17 by darksidelemm