projecthorus/radiosonde_auto_rx #539: error caused by config.py

icicmoi, 2021-07-06:
Hi all,
for information I find today the same problem by wanting to switch to version 1.5.4 as the one I encountered during the upgrade to v 1.5.3 .
[extract of the result of python3 config.py](https://bin.infini.fr/?352616f6f5552ec8#6hiL3CAZcmeWiBe47bnredvVyZPND9X1bgGQniUBxHqS)

However, I took care to check that oziexplorer is not validated (neither in station.cfg nor in config.py)

**Note:
for auto_rx to work correctly I had to comment out the entire ozi paragraph in the config.py file.**

I may have looked badly but I did not find any information on this subject in the doc.
That said, congratulations on the clarity of this documentation :-)
(she just made me tile again and I came to discover "git stash
"  :-)
And thanks for this open source software !

darksidelemm, 2021-07-06:
Sorry, i'm not sure if this is an issue or not? It suggests there was some issue in the configuration file rather than in the software itself...

icicmoi, 2021-07-11:
First of all thank you for taking the time to answer me, Mark ;-)

Le 06/07/2021 à 23:54, Mark Jessop wrote :
> Sorry, i'm not sure if this is an issue or not? It suggests there was some issue in the configuration file rather than in the software itself...

You are right and my apologies for my request:

I just understood / remembered that I systematically replaced my old "station.cfg" during an update; but you upgraded this file;)

73.
Philippe, f5rvx

closed 2021-07-11 by darksidelemm