projecthorus/radiosonde_auto_rx #540: Feature Request:  Option to support adhoc SSL

mdroberts1243, 2021-07-07:
Hi.  I would like to switch to using the Docker image, but I found that adding adhoc SSL is quite a bit of extra effort since I have to install lots of dependencies in order to pip install cryptography module.

I like to have external access to the Auto_Rx web page, but my router/proxy requires https.  So I've been making a one-line change to web.py:
```
def flask_thread(host="0.0.0.0", port=5000):
    """ Flask Server Thread"""
    socketio.run(app, host=host, port=port, ssl_context='adhoc')
```
This creates a requirement for the cryptography module in python.  On the native Pi version of Auto_Rx this isn't too big of an issue because it is a one-time installation and only the single line in web.py needs to be updated on each release.  But with the Docker image lots of things need to be installed (like gcc, build-essential, libssl-dev, libffi-dev, python-dev, and export CRYPTOGRAPHY_DONT_BUILD_RUST=1) before pip install cryptography.  This would have to be done on each release.

Is it possible to have ssl_context='adhoc' as an option, maybe in the configuration file for auto_rx?  Can the Docker image be created with cryptography already installed? 

Thanks,
-mark.

darksidelemm, 2021-07-07:
... interesting. I've tried the adhoc ssl mode in socketio before without success. Maybe it just didn't work in the version of socketio/flask that i was trying at the time. 

I just gave it a go on one of my stations, and while my browser indicated that the SSL cert was invalid, it just timed out after that. I then removed the ssl_context option and it worked as expected...

My flask / socketio versions are:
```
>>> socketio.__version__
'5.1.0'
>>> flask.__version__
'1.0.2'
>>> werkzeug.__version__
'0.14.1'
```

mdroberts1243, 2021-07-08:
@darksidelemm  Hi Mark.  I got adhoc to work in the Docker image by installing vim and all the dependencies listed above just using apt-get install and finally pip install cryptography (with the environment variable set to not build RUST).  It is working fine now, but would be a pain to do each update of the Docker image, hence the feature request.

I've run adhoc on my Pi version for a couple of years now, which lets me connect from outside the house.  On that system, I see:
>>> import socketio
>>> socketio.__version__
'5.2.1'
>>> import flask
>>> flask.__version__
'1.0.2'
>>> import werkzeug
>>> werkzeug.__version__
'0.14.1'
>>>

darksidelemm, 2021-07-09:
I'm still yet to get this functioning correctly on my system, and I'm wondering if it's socketio version related, but if you've been doing this for a while then it should work...

While I'm not against adding this as an option, there are other ways of putting a SSL layer in front of applications like auto_rx, and in ways where a real SSL certificate can be used instead of a self-signed cert. @snh can probably explain more.

mdroberts1243, 2021-11-01:
I'd certainly be interested in alternative ways of doing the reverse proxy through nginix.  @snh ?

I just upgraded to 1.5.6.  After the upgrade I had to add my SSL context. To do this I had to start a shell on the running docker image:
`docker exec -it radiosonde_auto_rx /bin/bash`

Then:
```
apt-get update
apt-get install vim
apt-get install gcc
apt-get install build-essential
apt-get install libssl-dev
apt-get install libffi-dev
apt-get install python-dev
export CRYPTOGRAPHY_DONT_BUILD_RUST=1
apt install python3-pip
pip3 install cryptography  ### May not be required after pip3
vim web.py ### change the flask thread to have the ssl context adhoc (see earlier in the thread)
```

After exiting the docker shell:
`docker restart radiosonde_auto_rx`