projecthorus/radiosonde_auto_rx #541: Autorx on raspberry

F6DKA, 2021-07-07:
After many installations following the protocol to the letter I can't display AutoRx on the browser. I always get the same error:
This site is inaccessible192.168.0.46 does not allow connection.
However I can access it and make it work without any problem with Putty. 
Can you help me please? 
thank you

fkemps, 2021-07-07:
Can you check if local port TCP:5000 is listening?

```
$ netstat -an | grep 5000
 or 
$ ss -splat | grep 5000
```
If there is a listening port then check your servers firewall settings. Might not allow inbound traffic to TCP port 5000.
You are btw very brief on the OS, platform versions etc you're installing......

My €2 cents,
Ferry

F6DKA, 2021-07-07:
***@***.***:~ $ netstat -an
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
tcp        0    256 192.168.0.46:22         192.168.0.39:25600
 ESTABLISHED
tcp6       0      0 :::22                   :::*                    LISTEN
udp        0      0 0.0.0.0:5353            0.0.0.0:*
udp        0      0 0.0.0.0:40465           0.0.0.0:*
udp        0      0 0.0.0.0:68              0.0.0.0:*
udp6       0      0 :::5353                 :::*
udp6       0      0 :::546                  :::*
udp6       0      0 :::54439                :::*
raw6       0      0 :::58                   :::*                    7
Active UNIX domain sockets (servers and established)
Proto RefCnt Flags       Type       State         I-Node   Path
unix  2      [ ]         DGRAM                    22564
 /run/user/1000/systemd/notify
unix  2      [ ACC ]     STREAM     LISTENING     22567
 /run/user/1000/systemd/private
unix  2      [ ACC ]     STREAM     LISTENING     22572
 /run/user/1000/gnupg/S.gpg-agent.extra
unix  2      [ ACC ]     STREAM     LISTENING     22573
 /run/user/1000/gnupg/S.gpg-agent.ssh
unix  2      [ ACC ]     STREAM     LISTENING     22574
 /run/user/1000/gnupg/S.dirmngr
unix  2      [ ACC ]     STREAM     LISTENING     22575
 /run/user/1000/gnupg/S.gpg-agent.browser
unix  2      [ ACC ]     STREAM     LISTENING     22576
 /run/user/1000/gnupg/S.gpg-agent
unix  2      [ ACC ]     STREAM     LISTENING     14512    /run/thd.socket
unix  2      [ ACC ]     STREAM     LISTENING     14515
 /run/avahi-daemon/socket
unix  2      [ ACC ]     STREAM     LISTENING     14517
 /var/run/dbus/system_bus_socket
unix  2      [ ACC ]     STREAM     LISTENING     14519
 /var/run/docker.sock
unix  3      [ ]         DGRAM                    10447
 /run/systemd/notify
unix  2      [ ACC ]     STREAM     LISTENING     10450
 /run/systemd/private
unix  2      [ ]         DGRAM                    10455
 /run/systemd/journal/syslog
unix  2      [ ACC ]     STREAM     LISTENING     10463
 /run/systemd/fsck.progress
unix  2      [ ACC ]     SEQPACKET  LISTENING     10465    /run/udev/control
unix  2      [ ACC ]     STREAM     LISTENING     10470
 /run/systemd/journal/stdout
unix  8      [ ]         DGRAM                    10472
 /run/systemd/journal/socket
unix  13     [ ]         DGRAM                    10488
 /run/systemd/journal/dev-log
unix  2      [ ACC ]     STREAM     LISTENING     14294
 /var/run/dhcpcd.sock
unix  2      [ ACC ]     STREAM     LISTENING     14295
 /var/run/dhcpcd.unpriv.sock
unix  2      [ ACC ]     STREAM     LISTENING     16781
 /run/containerd/containerd.sock.ttrpc
unix  2      [ ACC ]     STREAM     LISTENING     16784
 /run/containerd/containerd.sock
unix  2      [ ACC ]     STREAM     LISTENING     17557
 /var/run/docker/metrics.sock
unix  2      [ ACC ]     STREAM     LISTENING     17685
 /var/run/docker/libnetwork/d34d0eefe96c.sock
unix  2      [ ]         DGRAM                    17555    @00001
unix  2      [ ]         DGRAM                    15018
unix  2      [ ]         DGRAM                    10640
unix  3      [ ]         STREAM     CONNECTED     15537
 /run/systemd/journal/stdout
unix  3      [ ]         STREAM     CONNECTED     15467
[... 92 lines cut ...]
unix  3      [ ]         DGRAM                    22565
unix  3      [ ]         DGRAM                    22566
unix  3      [ ]         STREAM     CONNECTED     19596
unix  3      [ ]         STREAM     CONNECTED     22544
unix  2      [ ]         DGRAM                    22553
unix  2      [ ]         DGRAM                    22550
unix  3      [ ]         STREAM     CONNECTED     19597
***@***.***:~ $

 ss -splat
Total: 148
TCP:   3 (estab 1, closed 0, orphaned 0, timewait 0)

Transport Total     IP        IPv6
RAW       1         0         1
UDP       6         3         3
TCP       3         2         1
INET      10        5         5
FRAG      0         0         0

State     Recv-Q    Send-Q       Local Address:Port       Peer Address:Port
LISTEN    0         128                0.0.0.0:ssh             0.0.0.0:*
ESTAB     0         64            192.168.0.46:ssh        192.168.0.39:25600
LISTEN    0         128                   [::]:ssh                [::]:*
***@***.***:~ $

darksidelemm, 2021-07-07:
Given there's no entry in the list with :5000, auto_rx is not running.

Have you tried running up auto_rx manually, and does it give any errors? 
e.g. in the Initial Testing section: https://github.com/projecthorus/radiosonde_auto_rx/wiki#44-initial-testing

F6DKA, 2021-07-08:
Result scrip python3 auto_rx.py
$  python3 auto_rx.py
2021-07-08 08:46:32,090 INFO:Reading configuration file...
2021-07-08 08:46:32,099 WARNING:Config - Web Password not set, disabling
web control
2021-07-08 08:46:35,254 INFO:Config - Tested SDR #0 OK
2021-07-08 08:46:35,263 INFO:Started Flask server on http://0.0.0.0:5000
2021-07-08 08:46:35,264 INFO:Telemetry Logger - Started Telemetry Logger
Thread.
2021-07-08 08:46:35,272 INFO:OziMux - Started OziMux / Payload Summary
Exporter
2021-07-08 08:46:35,275 INFO:Sondehub Uploader - Started Sondehub Uploader
Thread.
2021-07-08 08:46:35,519 INFO:Version - Local Version: 1.5.4 - Up to date!
2021-07-08 08:46:35,520 INFO:Task Manager - SDR #0 has been allocated to
Scanner.
2021-07-08 08:46:36,128 INFO:Sondehub Uploader - Uploaded station
information to Sondehub.
2021-07-08 08:46:38,647 INFO:Scanner #0 - Starting Scanner Thread
2021-07-08 08:46:38,653 INFO:Scanner #0 - Running frequency scan.

darksidelemm, 2021-07-08:
And with auto_rx running like that can you access the web interface?

F6DKA, 2021-07-08:
no

darksidelemm, 2021-07-08:
OK, so with it running like that, try running (in another terminal on the rpi): `netstat -tnlp`
This will give similar output to the previous netstat command, but focused on TCP servers like what auto_rx is running.

F6DKA, 2021-07-08:
***@***.***:~/radiosonde_auto_rx/auto_rx $ netstat -tnlp
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
    PID/Program name
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN
     -
tcp6       0      0 :::22                   :::*                    LISTEN
     -

darksidelemm, 2021-07-08:
Was this running at the same time as you had auto_rx running in another terminal? (e.g. python3 auto_rx.py )
They have to be running simultaneously for this to be a meaningful test...

F6DKA, 2021-07-08:
there is another problem. if i change a parameter in station.cfg (for
example) i do a restart, the test command python3 auto_rx.py does not work
anymore....
***@***.***:~ $ python3 auto_rx.py
python3: can't open file 'auto_rx.py': [Errno 2] No such file or directory

I feel like I'm going to blow up the whole system .....

darksidelemm, 2021-07-08:
This is occurring because you are not in the radiosonde_auto_rx/auto_rx directory. You need to change into this directory first...

F6DKA, 2021-07-08:
ok

F6DKA, 2021-07-08:
ok it works... how to make AutoRx start in the right directory in case of
reboot ???

F6DKA, 2021-07-08:
the web page works only if the python3 command auto_rx.py is active in putty

darksidelemm, 2021-07-08:
If you set up auto_rx *as per the instructions* (e.g. follow https://github.com/projecthorus/radiosonde_auto_rx/wiki#451-option-1---operation-as-a-systemd-service-recommended ) then auto_rx will start up automatically. 

If you are not using the 'pi' user (as I suspect might be the case as you are **'ing out your username in your pastes), the you are going to need to do as the instructions say and modify the auto_rx.service file before enabling it, else it will not work.

F6DKA, 2021-07-08:
ok but which username are you talking about??? When I connect the Raspi I
use PI and the default password

darksidelemm, 2021-07-08:
If you are using the default pi username, then you don't need to edit the auto_rx.service file at all, and if you follow those instructions step by step it should result in auto_rx starting automatically on boot.

Once this is done, you can verify this is working by running: `sudo systemctl status auto_rx.service`

F6DKA, 2021-07-08:
***@***.***:~/radiosonde_auto_rx/auto_rx $ sudo systemctl status
auto_rx.service
Unit auto_rx.service could not be found.

darksidelemm, 2021-07-08:
Did you follow all the setup instructions at https://github.com/projecthorus/radiosonde_auto_rx/wiki#451-option-1---operation-as-a-systemd-service-recommended ?

In particular (assuming you are in the radiosonde_auto_rx/auto_rx directory):
```
$ sudo cp auto_rx.service /etc/systemd/system/
$ sudo systemctl enable auto_rx.service
$ sudo systemctl start auto_rx.service
```

F6DKA, 2021-07-08:
Ok everything seems to work now

fkemps, 2021-07-08:
....if not, you should get an "not following clear instructions award" ;-)
Good to see it progressed and auto_rx became running

F6DKA, 2021-07-08:
Thanks for everything and sorry ...

closed 2021-07-08 by darksidelemm